YosysHQ/yosys #3707: ice40: BRAM negedge read is one cycle late

niels-moller, 2023-03-19:
### Version

Yosys 0.26+50 (git sha1 0f2d226ae, gcc 10.2.1-6 -Og -fPIC)

### On which OS did this happen?

Linux

### Reproduction Steps

I'm working on a processor design, where the register file is read on negative clock edge (three read ports), and written on positive clock edge. Synthesized as 12 SB_RAM40_4KNR (since registers are 64 bits wide, and there's one copy for each read port). I've noticed that values read appear to be one cycle late. I expect the read values to be available and stable on positive clock edge (after some shallow combinational logic downstream of the BRAM read operation on previous negative edge), but it appears read values don't appear until the *second* positive edge after the negative edge when read address is available.

I'm attaching a minimized example (with an extra  .txt suffix because of github).

[top.v.txt](https://github.com/YosysHQ/yosys/files/11011546/top.v.txt)
[top.pcf.txt](https://github.com/YosysHQ/yosys/files/11011547/top.pcf.txt)

I synthesize this for ice40 and test on an "iCE40-HX8K Breakout Board". If I synthesize *without* BRAM, using 

`yosys -p 'synth_ice40 -noflatten -nobram -json top-ff.json' -f verilog top.v -L top-ff.ylog`

followed by 

`nextpnr-ice40 --hx8k --package ct256 --json top-ff.json --pcf top.pcf --asc top-ff.asc --log top-ff.nlog`

and load it with `icepack top-ff.asc top-ff.bin && iceprog top-ff.bin`

things work as expected: Leds light up in the pattern 11111001, and this also agrees with simulation using iverilog.

But if I synthesize in the intended way, striking the `-nobram` flag from the command line,
```
yosys -p 'synth_ice40 -noflatten -json top-bram.json' -f verilog top.v -L top-bram.ylog
nextpnr-ice40 --hx8k --package ct256 --json top-bram.json --pcf top.pcf --asc top-bram.asc --log top-bram.nlog
icepack top-bram.asc top-bram.bin
iceprog top-bram.bin
```

the leds instead lights up in the pattern 11100101. The leds are assigned two at a time on consecutive cycles, on positive clock edge, starting from the right, so the middle four bits shows that there's an additional delay cycle compared with synthesis using flipflops. It's as if there's an extra flipflop added on the BRAM read ports.

Not sure what further information is help for for tracking this down. next-pnr version is nextpnr-0.4-25-g445d3249. I'm also attaching the generated top-bram.json file (somewhat large, 17000 lines, but small compared to top-ff.json).
[top-bram.json.txt](https://github.com/YosysHQ/yosys/files/11011613/top-bram.json.txt), and the Makefile I use (in case I mistyped some of the commands above).
[Makefile.txt](https://github.com/YosysHQ/yosys/files/11011635/Makefile.txt)

Any advice is appreciated, and let me know what I can do to aid debugging.

### Expected Behavior

Leds on the board lighting up as 11111001

### Actual Behavior

Leds on the board lighting up as 11100101, indicating a cycle of extra delay.  (This unexpected behavior appears only if use of BRAM is enabled using synthesis).

gatecat, 2023-03-19:
Do you think it's possible the read is actually happening on the posedge not the negedge so it's appearing one cycle late?

We discovered the polarities were swapped on the up5k a year ago, maybe the hx8k is actually affected too?

https://github.com/YosysHQ/nextpnr/issues/910

niels-moller, 2023-03-19:
> Do you think it's possible the read is actually happening on the posedge not the negedge so it's appearing one cycle late?

Hmm, not sure, but I think that could explain the behavior I see. Is there some easy way to see if the generated json file is wrong (problem in yosys) or right (problem later, likely next-pnr)? Do the reverse engineered icestorm docs say if these config bits are the same or different for up5k and hx8k? 

I'm not familiar with these things at that level, but I guess it should be possible to pin-point the related bits in the .asc file to see what they are set to? If I try tools icebox_explain and icebox_asc2hlc, I see there are bits name "NegClk" in the .ramt_tile blocks, but not clear if that refers to read port or write port.

niels-moller, 2023-03-19:
According to https://github.com/YosysHQ/icestorm/blob/master/docs/ram_tile.html, "The NegClk bit in the RAMB tile negates the polarity of the WCLK port, and the NegClk bit in the RAMT tile negates the polarity of the RCLK port.", is that right? 

If I understand the output of icebox_explain, I get NegClk set on all 12 ramt tiles, and unset on the ramb tiles. Which looks correct to me, assuming the above docs are right for hx8k.

niels-moller, 2023-03-20:
I tried the below hack at the same place in next-pnr, and that appears to solve the problem, at least for this small example.
Should I file a separate issue (or pull request, but I'm not so comfortable proposing changes I barely understand) on the nextpnr repo?

[negclk.patch.txt](https://github.com/YosysHQ/yosys/files/11014943/negclk.patch.txt)

gatecat, 2023-03-20:
Thanks for your experiments here! I've done some experimenting with icecube and come to the conclusion that it's swapped for everything except the 1k devices.

See https://github.com/YosysHQ/nextpnr/pull/1128 and a docs update in https://github.com/YosysHQ/icestorm/pull/313.

Can you confirm it behaves correctly for you with this patch?

niels-moller, 2023-03-20:
myrtle ***@***.***> writes:

> Can you confirm it behaves correctly for you with this patch?

I'll try it out, but it may take a day or two.

-- 
Niels Möller. PGP key CB4962D070D77D7FCB8BA36271D8F1FF368C6677.
Internet email is subject to wholesale government surveillance.