YosysHQ/yosys #3713: Synthesis results can change based on comments or blank lines in Verilog source

algofoogle, 2023-03-25:
### Version

Yosys 0.22 (git sha1 f109fa3d4c5, gcc 8.3.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

# tl;dr

If you add just the right number of comments-only header lines (e.g. license text) or blank lines to the top of a Verilog file being run through yosys, you can find that in some cases it will change the synthesis result even though the rest of the file is unchanged.

This emerged through OpenLane, and I'm too new to yosys to know how to reproduce this completely with yosys alone, but hopefully I can still provide enough information or be guided to do so.

# Background

I encountered this using [efabless/caravel_user_project](https://github.com/efabless/caravel_user_project) (latest tag `mpw-9a`), via OpenLane, and reported it in https://github.com/The-OpenROAD-Project/OpenLane/issues/1697. Here's an example simple design intended for the sky130A PDK that demonstrates the issue:

[MYTEST.tar.gz](https://github.com/YosysHQ/yosys/files/11068620/MYTEST.tar.gz) - Contents:
```
verilog/rtl/MYTEST.v          # Top design; instantiates MYTEST_sub.
verilog/rtl/MYTEST_sub.v      # The file where blank lines change synthesis results!!
openlane/MYTEST/config.json   # Uses the above 2 Verilog files.
```

Using OpenLane with Caravel, we'd run `make MYTEST` and grab the synthesis result from `openlane/MYTEST/runs/MYTEST/*/results/synthesis/MYTEST.v`.

If you run the above `make MYTEST`, you'll get a certain result in `results/synthesis/MYTEST.v`.

If you add 20 blank lines to the top of the source `MYTEST_sub.v` file and run the make again, you'll get a **different** result in `results/synthesis/MYTEST.v`.

If you add 10 more blank lines (30 total) and run the make again, you'll get the **SAME** result as for only 20 blank lines.

If you add 10 more (40 total), you'll get a **DIFFERENT** result yet again.

Results for these 4 cases in my OpenLane environment are attached: 
[MYTEST_synth.tar.gz](https://github.com/YosysHQ/yosys/files/11068647/MYTEST_synth.tar.gz)

### Expected Behavior

Whitespace that doesn't actually alter the meaning of the Verilog code (especially adding blank lines or comment-only lines) should not change the synthesis result, allowing a consistent result regardless of whether comments (or blank lines) are used or not.

### Actual Behavior

It *seems* that the line numbers on which certain lines of Verilog code appear will have an effect on the yosys synthesis result, especially near a change in magnitude of the line number (i.e. 9 &rarr; 10, 99 &rarr; 100, 999 &rarr; 1000, etc).

algofoogle, 2023-03-25:
I would like to try this on Yosys `main`, but I need to work up to that yet. In the meantime, [I have a hypothesis](https://github.com/The-OpenROAD-Project/OpenLane/issues/1697#issuecomment-1483742941) that this is caused by Yosys internally naming certain things (e.g. registers?) based on the line number of where they are defined in the source Verilog, and if these names happen to be sorted (as plain strings) but the line numbers are not zero-padded, then ordering of things will change as you shift line numbers around.

algofoogle, 2023-03-25:
I've become aware of PR https://github.com/YosysHQ/yosys/pull/3665 and assuming my theory above is correct, maybe just zero-padding line/column numbers for now would be an easy way to get a little bit more predictability/consistency?

algofoogle, 2023-03-28:
Totally makes sense, @Ravenslofty. Definitely I can understand how shuffling names/ordering of cells could be expected to give different results. I was more tripped up by the fact that a file without comments produces different synthesis than exactly the same file *with* comments (especially given those comments are just a license, and some explanation, and not "magic markers" or whatever you call those comment-based hints/directives).

As for zero-padding, it's definitely just a hacky (but I assume quick) workaround and it misses several other cases, but at least padding to 5 chars each would mean a tiny increase in these internal strings, while maintaining line/column number readability for humans, and would allow 99,999 lines of Verilog in a single file (and each line could be up to 99,999 characters in length) while giving consistent results each time.

I've experimentally observed that I can achieve the same thing (at least for my relatively small files) by adding 1,000 blank lines to the start of my Verilog. That way, I know that all internal tags will already be padded to 4 digits... enough to cover up to 9,999 lines of code. In production, I won't do this... it was just to prove that it made the non-determinism introduced by comments/blanks go away.

erikcorry, 2026-10-23:
The way to fix this is two-fold.

Line numbers in identifiers should be zero-padded to 6 digits.  If you want to prevent indentation changes from affecting the output then column numbers should also be zero padded.  This ensures that when things are sorted the ordering doesn't change because comment lines were inserted.

The hash tables are almost insertion-ordered, but erase moves the last item into the deleted slot.  Some operations remove items in a hash-determined order so the order ends up depending on the hash, which is affected by comment lines.  Instead the erase operation needs to use tombstone entries.  These can accumulate, so you have to sometimes squeeze out the tombstones with a compaction, eg before the loops using cell_at() and wire_at().

I have a fix, but it's written by an LLM.