YosysHQ/yosys #3717: The `hierarchy -auto-top` command doesn't mark the top level module correctly

tilk, 2023-03-29:
### Version

0.27

### On which OS did this happen?

Linux

### Reproduction Steps

Take the following Verilog file:

```verilog
module m(input a, output b);
  n x(a, b);
endmodule

module n(input a, output b);
  assign b = !a;
endmodule
```

Load it into Yosys and run `hierarchy -auto-top`, for example:

```
yosys -p "hierarchy -auto-top" test.v
```

The interesting part is, the problem can be worked around by running certain commands before `hierarchy`, for example:

```
yosys -p "opt_clean; hierarchy -auto-top" test.v
```

Or:

```
yosys -p "write_verilog; hierarchy -auto-top" test.v
```

### Expected Behavior

Module `m` is detected as a top-level module:

```
2.3. Analyzing design hierarchy..
Top module:  \m
```

### Actual Behavior

Module `n` is incorrectly detected as a top-level module:

```
2.3. Analyzing design hierarchy..
Top module:  \n
```

This happens also for larger, real-world examples.

tilk, 2023-03-29:
The issue can be fixed by adding `design->sort()` to `Hierarchy::execute()` in `passes/hierarchy/hierarchy.cc`. As I'm unfamiliar with Yosys's design, I don't have any idea why this works.

arcturusannamalai, 2023-05-05:
problem with the code https://github.com/YosysHQ/yosys/blob/f93671eb8567f5548844e7065c808a5b4a32481d/passes/hierarchy/hierarchy.cc#L669 seems to be it does only one pass and picks the module containing the maximum instantiated cells not necessarily the top module.

the criterion for top module should be a combination of a high (not necessarily max) number of instances of modules as well as no self-instantiation (i.e. excluding test bench perhaps)

jix, 2023-05-23:
@arcturusannamalai That's not what `find_top_mod_score` does, the score is the maximum depth of any contained instance:

https://github.com/YosysHQ/yosys/blob/f93671eb8567f5548844e7065c808a5b4a32481d/passes/hierarchy/hierarchy.cc#L683

That is the correct metric to use here and taking the module with the highest score does result in a module that is not instantiated anywhere else (as otherwise the instantiating module would have a higher score).

The actual issues is that `find_top_mod_score` does not deal with abstract (`$abstract...`) modules and so doesn't see the module instances. Files specified on the command line or read with the `read` command are loaded using `read_verilog` in `-defer` mode which uses abstract modules even for non-generic modules.

ldoolitt, 2024-11-06:
I found my problem.  Woo-hoo!
I suggest adding the following patch to help the next poor slob whose code base accidentally has a stray (* top *) attribute where it wasn't expected.
```
diff --git a/passes/hierarchy/hierarchy.cc b/passes/hierarchy/hierarchy.cc
index a8d7d5a53..34994957e 100644
--- a/passes/hierarchy/hierarchy.cc
+++ b/passes/hierarchy/hierarchy.cc
@@ -1003,8 +1003,10 @@ struct HierarchyPass : public Pass {
 
                if (top_mod == nullptr)
                        for (auto mod : design->modules())
-                               if (mod->get_bool_attribute(ID::top))
+                               if (mod->get_bool_attribute(ID::top)) {
                                        top_mod = mod;
+                                       log("attribute 'top' found and respected for module %s\n", log_id(mod));
+                               }
 
                if (top_mod == nullptr)
                {
```

nakengelhardt, 2024-11-06:
Thanks for the reminder, this issue was indeed fixed with #4130.

closed 2024-11-06 by nakengelhardt