YosysHQ/yosys #3724: Extract_fa not extracting all full adders

Sysynth, 2023-04-05:
### Version

0.27

### On which OS did this happen?

Linux

### Reproduction Steps

Hi,

I am trying to synthesize a simple 4-bit ripple carry adder. Since abc does not support multi-output cells, it is needed to first extract the full adders from the design and then custom map them to my single bit full adder cell using the `techmap` command.

However, the problem is that, for some reason, the command `extract_fa` is only extracting 3 of the 4 full adders.

This is my Yosys script:

```
read_verilog full_adder.v

hierarchy -check -top full_adder

read_liberty -lib sky130_fd_sc_hd__tt_025C_1v80.lib

proc
opt_clean

alumacc
opt -fast

techmap
opt -fast
extract_fa -fa -v
techmap -map OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v
opt -fast

abc -liberty sky130_fd_sc_hd__tt_025C_1v80.lib
opt -fast

check

write_verilog synth_fa.v

stat -liberty sky130_fd_sc_hd__tt_025C_1v80.lib
```

Is there anything wrong in my script that could be causing the issue?

Thanks in advance.

### Expected Behavior

Same module synthesized with another tool. 

```
module full_adder ( a, b, c_in, c_out, sum );
  input [3:0] a;
  input [3:0] b;
  output [3:0] sum;
  input c_in;
  output c_out;
  wire   n1, n2, n3;

  sky130_fd_sc_hd__fa_1 U1 ( .A(b[3]), .B(a[3]), .CIN(n1), .COUT(c_out), .SUM(
        sum[3]) );
  sky130_fd_sc_hd__fa_1 U2 ( .A(b[2]), .B(a[2]), .CIN(n2), .COUT(n1), .SUM(
        sum[2]) );
  sky130_fd_sc_hd__fa_1 U3 ( .A(b[1]), .B(a[1]), .CIN(n3), .COUT(n2), .SUM(
        sum[1]) );
  sky130_fd_sc_hd__fa_1 U4 ( .A(c_in), .B(b[0]), .CIN(a[0]), .COUT(n3), .SUM(
        sum[0]) );
endmodule
```

### Actual Behavior

Only uses 3 full adder cells. Instead of the fourth one, it needs 6 other cells.
 
```
(* top =  1  *)
(* src = "../../full_adder.v:1.1-10.10" *)
module full_adder(a, b, c_in, c_out, sum);
  wire _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  (* force_downto = 32'd1 *)
  (* src = "../../full_adder.v:8.24-8.36|/usr/local/bin/../share/yosys/techmap.v:274.23-274.25" *)
  wire [4:0] _04_;
  (* src = "../../full_adder.v:1.33-1.34" *)
  input [3:0] a;
  wire [3:0] a;
  (* src = "../../full_adder.v:2.33-2.34" *)
  input [3:0] b;
  wire [3:0] b;
  (* src = "../../full_adder.v:3.27-3.31" *)
  input c_in;
  wire c_in;
  (* src = "../../full_adder.v:4.32-4.37" *)
  output c_out;
  wire c_out;
  (* src = "../../full_adder.v:5.38-5.41" *)
  output [3:0] sum;
  wire [3:0] sum;
  sky130_fd_sc_hd__nand2_1 _05_ (
    .A(b[3]),
    .B(a[3]),
    .Y(_03_)
  );
  sky130_fd_sc_hd__a21oi_1 _06_ (
    .A1(a[2]),
    .A2(_04_[1]),
    .B1(b[2]),
    .Y(_00_)
  );
  sky130_fd_sc_hd__o22ai_1 _07_ (
    .A1(b[3]),
    .A2(a[3]),
    .B1(_04_[1]),
    .B2(a[2]),
    .Y(_01_)
  );
  sky130_fd_sc_hd__o21ai_0 _08_ (
    .A1(_00_),
    .A2(_01_),
    .B1(_03_),
    .Y(c_out)
  );
  sky130_fd_sc_hd__xnor2_1 _09_ (
    .A(b[3]),
    .B(a[3]),
    .Y(_02_)
  );
  sky130_fd_sc_hd__xnor2_1 _10_ (
    .A(_04_[2]),
    .B(_02_),
    .Y(sum[3])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v:17.28-17.75" *)
  sky130_fd_sc_hd__fa_1 _11_ (
    .A(a[2]),
    .B(b[2]),
    .CIN(_04_[1]),
    .COUT(_04_[2]),
    .SUM(sum[2])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v:17.28-17.75" *)
  sky130_fd_sc_hd__fa_1 _12_ (
    .A(a[1]),
    .B(b[1]),
    .CIN(_04_[0]),
    .COUT(_04_[1]),
    .SUM(sum[1])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v:17.28-17.75" *)
  sky130_fd_sc_hd__fa_1 _13_ (
    .A(a[0]),
    .B(b[0]),
    .CIN(c_in),
    .COUT(_04_[0]),
    .SUM(sum[0])
  );
  assign _04_[4:3] = { 1'h0, c_out };
endmodule

```

Ravenslofty, 2023-04-06:
What's your input Verilog, i.e. `full_adder.v`

tristanitschner, 2023-04-09:
As a workaround you could use this, it infers properly:

```verilog
module adder(
	input  wire a,
	input  wire b,
	input  wire c_in,
	output wire c_out,
	output wire sum
);

assign {c_out, sum} = a + b + c_in;

endmodule

module full_adder #(
	parameter width = 4
) ( 
	input  wire [width-1:0] a,
	input  wire [width-1:0] b,
	input  wire             c_in,
	output wire             c_out,
	output wire [width-1:0] sum
);

wire [width:0] carry;

assign carry[0] = c_in;

genvar i;
generate for (i = 0; i < width; i = i + 1) 
	adder adder_inst(
		.a     (a[i]),
		.b     (b[i]),
		.c_in  (carry[i]),
		.c_out (carry[i+1]),
		.sum   (sum[i])
	);
endgenerate

assign c_out = carry[width];

endmodule

```

Ravenslofty, 2023-04-09:
Honestly, we usually advise against using `extract_fa` for this: as a rule of thumb, going from a more abstract cell to a more concrete cell is easy, and going the other way is hard.

With that in mind, the general approach for this is along the lines of @tristanitschner's suggestion, but using Yosys internals instead.

Yosys sees a `+` in the input Verilog and produces a `$add` cell, and flows then subsequently call the `alumacc` pass which transforms `$add` into `$alu` cells (along with a lot of other arithmetical operations, which is why it's best to work with `$alu` rather than `$add`).

You can then write a Verilog file for `techmap` along the lines of [this] (https://github.com/YosysHQ/yosys/blob/master/techlibs/intel_alm/common/arith_alm_map.v) but replacing `MISTRAL_ALM_ARITH` with the equivalent instantiation of `sky130_fd_sc_hd__fa_1` instead, and call `techmap -map your_file.v`.

By doing this you get the advantage of directly being mapped into the hard cells you want, while avoiding the need for direct instantiation that @tristanitschner's solution needs.

Having explained all of this, I should point out that by default Yosys will turn a `$alu` cell into a Brent-Kung adder that is likely faster than the ripple carry cells.

Sysynth, 2023-04-10:
I understand it now, thank you both for your help!

And thank you, @Ravenslofty, for the detailed explanation. :)

Marking this as closed.

closed 2023-04-10 by Sysynth