YosysHQ/yosys #3731: 0.28: Tests fail: Test ../../techlibs/ecp5/cells_sim.v ->../../techlibs/ecp5/cells_sim.v:448: syntax error

yurivict, 2023-04-16:
### Version

0.28

### On which OS did this happen?

BSD

### Reproduction Steps

Rn tests

### Expected Behavior

n/a

### Actual Behavior

```
Test ../../techlibs/achronix/speedster22i/cells_sim.v -> ok
Test ../../techlibs/anlogic/cells_sim.v -> ok
Test ../../techlibs/coolrunner2/cells_sim.v -> ok
Test ../../techlibs/ecp5/cells_sim.v ->../../techlibs/ecp5/cells_sim.v:448: syntax error
../../techlibs/ecp5/cells_sim.v:448: error: invalid module item.
gmake[1]: *** [Makefile:855: test] Error 2
gmake[1]: Leaving directory '/usr/ports/cad/yosys/work/yosys-yosys-0.28'
*** Error code 1
```

Luflosi, 2023-04-16:
I'm seeing the same error on Linux.

InternetUnexplorer, 2023-04-17:
I bisected it to 266f81816bbede3f42fabe59e7d0cd8873e1a467, but in hindsight I should've just looked at the error first because it says what the problem is; for some reason iverilog is unhappy with the `$error` on line 448 of `techlibs/ecp5/cells_sim.v`. I noticed NixOS has a version of iverilog that is over 2 years old (v11.0), perhaps this is a bug with iverilog that has been fixed since then? But I haven't had time to check yet.

Luflosi, 2023-04-17:
I'll try updating iverilog to the latest Git version and see if that fixes the problem.

yurivict, 2023-04-17:
I used iverilog-11.0

Luflosi, 2023-04-17:
I didn't yet manage to run the iverilog test suite after updating it but yosys compiles fine and its test suite runs without errors.

Luflosi, 2023-04-18:
I just found out that iverilog has a 12.0 git tag (not GutHub release) that I expect to work but I haven't managed to compile it on Linux yet.

Luflosi, 2023-05-07:
I can confirm that everything works as expected with iverilog 12.

Luflosi, 2023-05-10:
As far as I'm concerned, this issue can be closed now.

closed 2023-05-10 by gatecat