YosysHQ/yosys #3743: LUTs not topologically sorted after write_verilog

cgouert, 2023-04-26:
### Version

Yosys 0.9+2406 (git sha1 334ec5fa, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Given this verilog program (adder.v)

```
module adder(G11,G12,G14);
input [7:0] G11,G12;
output [7:0] G14;
assign G14 = G11+G12;
endmodule
```

Yosys script: `read_verilog adder.v; synth; abc -lut 6; write_verilog -noexpr -v adder_out.v` 

LUTs are not topologically sorted in adder_out.v

However, executing `read_verilog adder.v; synth; abc -g simple,-MUX; write_verilog -noexpr -v adder_out.v`  yields topologically sorted logic gates.

cc: @jimouris

### Expected Behavior

I would expect the LUTs to be topologically sorted in the generated Verilog.

### Actual Behavior

LUTs are not topologically sorted when dumped to a Verilog netlist.

whitequark, 2023-04-29:
There's no guarantee that the generated Verilog is topologically sorted. What you see when using it with `abc` is a coincidence.

cgouert, 2023-05-03:
Thank you for your reply; I can also replicate this behavior with other circuits, but this may be a coincidence as you mentioned. I know that Yosys has the capability to topologically sort circuits with the torder command  (which prints a topological ordering of the nodes/gates). Is there any way to combine this somehow with the write_verilog command to reflect this in the generated netlist?

whitequark, 2023-05-05:
Not that I know of.

jimouris, 2023-05-12:
It seems that it is not a coincidence: https://github.com/berkeley-abc/abc/issues/60 and `abc` sorts the gates. It seems that it's a bug on their end but can also be resolved on yosys' side with the output of the `torder` command.

whitequark, 2023-05-12:
Sorry, I was unclear. By "coincidence" I meant that, as far as I know, Yosys makes no guarantee that the Verilog output (or any other output) will be topologically sorted after running `abc`.

arcturusannamalai, 2023-05-21:
my attempt at a solution https://github.com/YosysHQ/yosys/pull/3774

closed 2023-05-21 by cgouert