YosysHQ/yosys #3768: Netlist for decoder

mistralino, 2023-05-18:
### Version

Yosys 0.28+12 (git sha1 266036c6f, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux, Windows

### Reproduction Steps

Creating a netlist of the following code using Yosys 0.28+12 (git sha1 266036c6f, clang 10.0.0-4ubuntu1 -fPIC -Os):

module decoder3_8(y, a);

  output reg [7:0] y;
  input wire [2:0] a;

  always @(*)  // no storage in combinational logic, 
                         // so each time any input changes, 
                        // the output may be affected
    case (a)
      3'b000: y = 8'b00000001;
      3'b001: y = 8'b00000010;
      3'b010: y = 8'b00000100;
      3'b011: y = 8'b00001000;
      3'b100: y = 8'b00010000;     
      3'b101: y = 8'b00100000;
      3'b110: y = 8'b01000000;
      3'b111: y = 8'b10000000;
    endcase

endmodule // decoder3_8

results in:

[image]

This does not seem to make much sense. In contrast, using Yosys 0.9 (git sha1 1979e0b1, i686-w64-mingw32.static-g++ 5.5.0 -Os) results in the following netlist:

[image]

which looks fine.

### Expected Behavior

The netlist created by Yosys 0.9 (git sha1 1979e0b1, i686-w64-mingw32.static-g++ 5.5.0 -Os), see above

### Actual Behavior

The netlist created by Yosys 0.28+12 (git sha1 266036c6f, clang 10.0.0-4ubuntu1 -fPIC -Os), which does not make much sense

nakengelhardt, 2023-05-23:
That's a desired optimization added in #3315: mapping this kind of look-up table to ROM is often a more efficient implementation (and when it is not, it will subsequently be mapped back to logic by memory_dff). You can disable it by calling `proc` with `-norom`.

mistralino, 2023-05-23:
Thanks for the quick response ;-)

closed 2023-06-30 by nakengelhardt