YosysHQ/yosys #3772: frontend support non-synthesizable constructs in Verilog

ctormuthu, 2023-05-18:
### Feature Description

Currently if code has display or finish etc in testbench or the initial blocks of a module we abort the AST processing of the Verilog; I request that tool be more forgiving if these offending, simulation only, constructs are present in the initial block.

e.g.

```verilog
module nonblock();
    reg x , y, z ;
    // Nonblocking assignments
initial begin
    $display("x y z at time %0t", $time);
    $monitor("%d %d %d at time %0t", x, y, z, $time);
    x <=  #10  1'b1;// The simulator assigns 1 to x at time 10
    y <=  #20  1'b0;// The simulator assigns 0 to y at time 20
    z  <=  #40  1'b1;// The simulator assigns 1 to z at time 40
end
endmodule
```
this design will not be loaded unless we comment out the '$display'

Ravenslofty, 2023-05-20:
`ifndef SYNTHESIS exists to section off non-synthesisable Verilog.

closed 2023-05-22 by ctormuthu

whitequark, 2023-05-23:
We could support `$display` and `$monitor` statements though (and there is ongoing work to do so).