YosysHQ/yosys #3773: Yosys synthesis breaks CPU functionality

flaviens, 2023-05-20:
### Version

Yosys 0.29+11 (git sha1 acfdc5cc4246 gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed that for one specific CPU written in SystemVerilog and whose pickled version is in the following archive, the CPU behaves incorrectly for some inputs, _only_ when it passes through Yosys, without any expected functional transformation.

[cva6_vanilla.zip](https://github.com/YosysHQ/yosys/files/11522073/cva6_vanilla.zip)

The Yosys script is the following (I also tried without the `-sv` for the back-end, without better success):
```
yosys read_verilog -defer -sv $VERILOG_INPUT
yosys hierarchy -top $TOP_MODULE -check
yosys proc
yosys opt -purge
yosys write_verilog -sv -noattr $VERILOG_OUTPUT
```

The problem does not seem to come from sv2v, since the output of sv2v still behaves correctly.

Here are the waveforms:
[waveforms.zip](https://github.com/YosysHQ/yosys/files/11522085/waveforms.zip)

The RISC-V ELF file running is the following:
[app_buggy.zip](https://github.com/YosysHQ/yosys/files/11522094/app_buggy.zip)

It seems that the first instruction that behaves incorrectly is the  `fmadd.d	f3,f10,f4,f6,rup` at PC=0x8001f7e8`.

### Expected Behavior

We expect the CPU to write the value `0x7ff80..0` to the SRAM address `0x3`.

[image]

### Actual Behavior

The CPU writes `0x7ff80..0` instead.

Ravenslofty, 2023-05-20:
Are you sure about the "actual behaviour" section here? Doesn't that just match what the expected behaviour section was?

flaviens, 2023-05-20:
Yes sorry, copy pasted too much :) fixed

flaviens, 2023-05-20:
For your convenience, I created a Docker image associated with [this repo](https://github.com/flaviens/cva6_yosys_issue).

You can reproduce the issue with:
```
docker pull docker.io/ethcomsec/cva6-yosys-issue
docker run -it ethcomsec/cva6-yosys-issue

make -C cva6_yosys_issue run_instrumented # Test that fails after passing through Yosys
make -C cva6_yosys_issue run_vanilla # Test that succeeds, using the design without passing it through the Yosys script
```

flaviens, 2023-05-28:
Hi! Have you had a chance to look at it already? Let me know if you need more info.
Thanks!

whitequark, 2023-05-28:
It is significantly more likely that someone will take the time to look at your issue in a timely way if you provide a reduced test case.

flaviens, 2023-05-28:
Hi @whitequark , thank you very much for your response! I understand your point. Unfortunately, since it's a CPU (and I'm not one of its maintainers at all)  it's difficult to break it apart and to build test cases for its subpieces. I was hoping it could be related to a problem that you already know of, since all mentioned experiments suggest that the issue is inherent to Yosys.
Thanks!

whitequark, 2023-05-28:
Ah, I somehow missed that you were not the author of the CPU. Yes, that is difficult to reduce, I agree. Unfortunately resources to do so are limited everywhere.

KatCe, 2023-10-05:
Hello. I have been able to locate the bug in the example above and have generated a tiny test case that reproduces it. Please find it here: https://github.com/KatCe/yosys_issue_3773 . Yosys transforms a Verilog function into a wrong constant. It can be seen in the code (see description in README), so no need to simulate. Hope that helps to solve the issue.

Ravenslofty, 2023-11-02:
This got brought up in the Team JF, so I thought I'd take a look.

The TCL script can at least be minimised a little further:
```tcl
if { [info exists ::env(VERILOG_INPUT)] }    { set VERILOG_INPUT $::env(VERILOG_INPUT) }       else { puts "Please set VERILOG_INPUT environment variable"; exit 1 }
if { [info exists ::env(VERILOG_OUTPUT)] }   { set VERILOG_OUTPUT $::env(VERILOG_OUTPUT) }     else { puts "Please set VERILOG_OUTPUT environment variable"; exit 1 }
if { [info exists ::env(TOP_MODULE)] }       { set TOP_MODULE $::env(TOP_MODULE) }             else { puts "Please set TOP_MODULE environment variable"; exit 1 }

yosys -import

read_verilog -defer -sv $VERILOG_INPUT
hierarchy -top $TOP_MODULE -check
proc_mux
proc_dff
proc_clean
opt_clean -purge

write_verilog -sv -noattr $VERILOG_OUTPUT
```

which amounts to breaking down the `proc` and `opt` passes into the subpasses that actually have an effect here.

Ravenslofty, 2023-11-02:
**I am not going to claim to be a language lawyer**; there's @clairexen and @zachjs for that, but what I think is happening is this:

IEEE 1800-2012 11.4.3:
> If neither operand of the power operator is real, then the result type shall be determined as outlined in 11.6.1
and 11.8.1.

IEEE 1800-2012 11.6.1:
> Table 11-21 shows how the form of an expression shall determine the bit lengths of the results of the
expression. In Table 11-21, i, j, and k represent expressions of an operand, and L(i) represents the bit
length of the operand represented by i.

IEEE 1800-2012 Table 11-21 (edited for brevity):

| Expression | Bit length | Comments |
|------------|-----------|-------------|
| i op j, where op is: `>>` `<<` `**` `>>>` `<<<` | L(i) | j is self-determined |

in `tiny_direct.sv`:
```sv
module tiny_top (
        result_o
);
  output wire [63:0] result_o;
        assign result_o = 2 ** (52 - 1);
endmodule
```
the expression `2 ** (52 - 1)` is promoted to `64'd2 ** (52 - 1)` because of the `assign`, and evaluated at 64-bit precision, getting `64'h0008000000000000`.

In `tiny_reg_function.sv`:
```sv
module tiny_top (
        result_o
);

        output wire [52-1:0] result_o;
        function automatic [52 - 1:0] sv2v_cast_14681;
                input reg [52 - 1:0] inp;
                sv2v_cast_14681 = inp;
        endfunction

  reg [52-1:0] test;

  assign test = 2 ** (52 - 1);

        assign result_o = sv2v_cast_14681(test);
endmodule
```

the expression `2 ** (52 - 1)` is promoted to `52'd2 ** (52 - 1)` because it's on the right hand side of a `reg` initialiser, and evaluated at 52-bit precision, getting `52'h8000000000000`.

In `tiny_function.sv`:
```sv
module tiny_top (
        result_o
);

        output wire [52-1:0] result_o;
        function automatic [52 - 1:0] sv2v_cast_14681;
                input reg [52 - 1:0] inp;
                sv2v_cast_14681 = inp;
        endfunction

        assign result_o = sv2v_cast_14681(2 ** (52 - 1));
endmodule
```

there is nothing to promote the width of the `2`, so it gets evaluated at the default width of 32 bits, getting `32'h0` because the expression is too large, which is then zero-extended to 52-bit-wide for the function parameter.

Again, I am not a language lawyer, but I think this might be how Yosys is interpreting the design, regardless of correctness: Yosys does not miscompile `tiny_function.sv` if `2` is replaced with `52'd2`.

zachjs, 2023-11-03:
@Ravenslofty What you've shown above is a bug in the Yosys frontend. All three examples should indeed behave the same. Function calls are "assignment-like contexts". This means values passed to functions should use width of the function arg they are passed to. I think before simplifying AST_FCALL args, we'd need to figure out the width of corresponding declaration in the function.