YosysHQ/yosys #3805: DLatches do not synthesize in AIGER

Pflyg, 2023-06-12:
### Version

0.24

### On which OS did this happen?

Linux

### Reproduction Steps

```
module mux (
  input wire select_1,
  input wire select_0,
  input wire in_0,
  input wire in_1,
  input wire in_2,
  input wire in_3,
  input clk,
  output out
);

always @(*) begin
  if (!select_1 && !select_0) begin
    out = in_0;
  end else if (!select_1 && select_0) begin
    out = in_1;
  end else if (select_1 && !select_0) begin
    out = in_2;
  end else if (select_1 && select_0) begin
    out = in_3;
  end
end

endmodule
```
```

read_verilog mux.vl;
# Sets the top module and overrides parameters
hierarchy -top mux;
proc;
check;
wreduce;
alumacc;
fsm;
memory -nomap;
memory_map;
techmap;

abc -fast;
hierarchy -check;
stat;
check;

# Maps the internal representation to aiger
aigmap;
write_aiger -ascii -symbols impl.aag;
```

### Expected Behavior

The above code should synthesize to valid AIGER code.

### Actual Behavior

The code cannot synthesize because there are D-Latches present:
```
17. Executing AIGER backend.
ERROR: Unsupported cell type: $_DLATCH_N_ ($auto$ff.cc:266:slice$230)
```

gatecat, 2023-06-12:
As a workaround, you can replace `abc -fast` with just `abc` (which then does enough logic optimisation to realise the latch is always enabled) and then add an `opt -fine` call before `aigmap` (which replaces the always-enabled latch with a wire).

Pflyg, 2023-06-12:
@gatecat
Thanks for the tip, seems to work well for now