YosysHQ/yosys #3807: Synthesis bug or maybe wrong understanding

BFH-ktt1, 2023-06-16:
### Version

Yosys 0.30+16

### On which OS did this happen?

Linux

### Reproduction Steps

```
    case (s_sevenSegSelReg)
      3'd4    : begin
//                  nSevenSeg <= { {6{1'b1}}, {2{toggleReg}} };
                  nSevenSeg[7:2] <= 6'b111111;
                  nSevenSeg[1]   <= toggleReg;
                  nSevenSeg[0]   <= toggleReg;
                end
      3'd1    : nSevenSeg <= ~s_segs3;
      3'd2    : nSevenSeg <= ~s_segs2;
      3'd3    : nSevenSeg <= ~s_segs1;
      default : nSevenSeg <= 8'hFF;
    endcase
```
In this verilog description the marked-out line does not reproduce the same behavior as the "alternative" below it.

### Expected Behavior

I would expect that both versions produce the same hardware, but they do not the marked out line produces a different implementation as the not-marked out lines.

### Actual Behavior

When I use the not-marked out lines I get the expected behavior, however, when I use the marked-out line I get following behavior:
```
nSevenSeg[7:1] <= 7'b0000000;
nSevenSeg[0] <= toggleReg;
```
I attached the complete module for debugging.
[rgbarray.zip](https://github.com/YosysHQ/yosys/files/11772835/rgbarray.zip)

zachjs, 2023-06-17:
At least part of the problem is that `toggleReg` has width 3, not 1. This means that replicating it twice has width 6, not 2 as seems to be intended.

BFH-ktt1, 2023-06-18:
@zachjs : You're completely right, I completely overlooked that. I fixed it and now it is working as expected. Sorry for the inconvenience, my bat :-) Thanks for the great work!

closed 2023-06-18 by BFH-ktt1