YosysHQ/yosys #3814: Yosys fails to parse "endmodule;" due to a semicolon

yurivict, 2023-06-22:
### Version

Yosys 0.30+1 (git sha1 5813809ad, gcc 12.2.0 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

obvious

### Expected Behavior

obvious

### Actual Behavior

Failure due to a semicolon.

Not sure if this is a Verilog vs. SystemVerilog issue, but semicolon is natural to occur at this location, and Yosys would be more robust if it was passed by Yosys.

jix, 2023-06-23:
Neither the Verilog nor the SystemVerilog standard allow a semicolon there.

yurivict, 2023-06-23:
> Neither the Verilog nor the SystemVerilog standard allow a semicolon there.

Ok. But Fusion Compiler though accepts semicolons there. Maybe adding a bit of flexibility there wouldn't hurt.

I've ran Yosys on scripts that were normally compiled by the Fusion Compiler and noticed this problem.