YosysHQ/yosys #3820: CXXRTL: incorrect result of shl operator

YikeZhou, 2023-06-24:
### Version

Yosys 0.30+33 (git sha1 f9257d319, gcc 8.3.0-6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Verilog file `top.v`:
```verilog
module top (
    output [5:0] o,
    input clk,
    input [2:0] i1,
    input [2:0] i2
  );
  wire [2:0] shl_res;
  reg [2:0] gt_res;
  assign o = {shl_res, gt_res};
  assign shl_res = (i2 <<< i1);
  always @(posedge clk)
    gt_res <= (i1 > shl_res);
endmodule
```

C++ file `replay.v`:
```cpp
#include "top.cpp"
#include <iostream>

int main()
{
	cxxrtl_design::p_top top;
	cxxrtl::debug_items items;
	top.debug_info(items);

	top.step();

	// Set inputs
	top.p_i2.set<uint8_t>(4u);
	top.p_i1.set<uint8_t>(1u);

	top.p_clk.set(false);
	top.step();
	top.p_clk.set(true);
	top.step();
	top.debug_eval();

	// Check outputs
	std::cout << "shl_res = " << top.p_shl__res.get<uint32_t>() << "\n";
	std::cout << "gt_res = " << top.p_gt__res.get<uint32_t>() << "\n";
	std::cout << "o = " << top.p_o.get<uint32_t>() << "\n";

	return 0;
}
```

Run:
```
yosys -p "read_verilog top.v; write_cxxrtl top.cpp"
g++ replay.cpp -I$(yosys-config --datdir)/include
./a.out
```

### Expected Behavior

`shl_res` should have been truncated to 3-bit.
```
shl_res = 0
gt_res = 1
o = 1
```

### Actual Behavior

```
shl_res = 8
gt_res = 0
o = 64
```

whitequark, 2024-01-05:
Still an issue on current master.

whitequark, 2024-01-20:
This was fixed by commit ff53f3d2b6e267358cf4c3d08574416c79441ef9. Thank you for the report!

closed 2024-01-20 by whitequark