YosysHQ/yosys #3821: CXXRTL: bus error when performing the ">>>" operation

YikeZhou, 2023-06-24:
### Version

Yosys 0.30+33 (git sha1 f9257d319, gcc 8.3.0-6 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Verilog file `top.v`:
```verilog
module top (
    output reg [2:0] shr,
    input clk,
    input wire signed [2:0] i1,
    input wire signed [2:0] i2
  );
  always @(posedge clk)
    shr <= (i1 >>> i2);
endmodule
```

C++ file `replay.cpp`:
```cpp
#include "top.cpp"
#include <iostream>

int main()
{
	cxxrtl_design::p_top top;
	top.step();

	// Set inputs
	top.p_i1.set<uint8_t>(4u);
	top.p_i2.set<uint8_t>(4u);

	top.p_clk.set(false);
	top.step();
	top.p_clk.set(true);
	top.step();

	// Check outputs
	std::cout << "shr = " << top.p_shr.get<uint32_t>() << "\n";

	return 0;
}
```

Run:
```
yosys -p "read_verilog top.v; write_cxxrtl top.cpp"
g++ replay.cpp -I$(yosys-config --datdir)/include
./a.out
```

### Expected Behavior

The program should print the value of `shr` and exit.

### Actual Behavior

The program exited unexpectedly due to a "bus error":
```
Program received signal SIGBUS, Bus error.
```
at
https://github.com/YosysHQ/yosys/blob/f9257d319271917aac30f7bfb4f5b154ff559b18/backends/cxxrtl/cxxrtl.h#L450

whitequark, 2024-01-05:
Still an issue on current master.

whitequark, 2024-01-19:
> Still an issue on current master.

Actually, I take that back, I wasn't able to reproduce this on either 0.36 or 0.37.

whitequark, 2024-01-20:
This was fixed in commit ded63bedd589255e6e06747edcdbf262444ac379. Thank you for the report!

closed 2024-01-20 by whitequark