YosysHQ/yosys #3831: FSM Extraction does not work

myzinsky, 2023-07-05:
### Version

Yosys 0.30+48 (git sha1 14d50a176, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

Followed the steps in this stack overflow answer:

https://stackoverflow.com/questions/32645965/fsm-export-using-yosys 

Using oss-cad-suite

### Expected Behavior

KISS-File of FSM-Structure shall be generated

### Actual Behavior

The extraction step passes, however, there is no kiss-file generated.

nakengelhardt, 2023-07-05:
Can you post the full design and script to reproduce this?

myzinsky, 2023-07-05:
It is in the link above:

test.v

```verilog
module test(input clk, rst, ctrl, output [3:0] O);
    reg [1:0] state;
    always @(posedge clk) begin
        O <= 0;
        if (rst) begin
            state <= 0;
        end else case (state)
            0: begin
                state <= ctrl ? 1 : 2;
                O <= 1;
            end
            1: begin
                O <= 2;
                if (ctrl) begin
                    state <= 2;
                    O <= 3;
                end
            end
            2: begin
                O <= 4;
                if (ctrl) begin
                    state <= 3;
                    O <= 5;
                end
            end
            3: begin
                if (!ctrl)
                    state <= 2'b00;
            end
        endcase
    end
endmodule
```

Yosys commands:

```
read_verilog test.v
proc; opt; fsm -nomap
fsm_export -o test.kiss2
```

nakengelhardt, 2023-07-05:
The script is wrong; `fsm` requires the previous `opt` to be called with `-nosdff -nodffe`. It works with this script:
```
read_verilog test.v
proc
opt -nodffe -nosdff
fsm -nomap
fsm_export -o test.kiss2
```

closed 2023-07-05 by nakengelhardt