YosysHQ/yosys #3832: Extraction of FSM from VHDL

myzinsky, 2023-07-05:
### Version

Yosys 0.30+48 (git sha1 14d50a176, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

This example State-Machine in VHDL (fsm.vhd):

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity fsm is
    Port (
        clk   : in  STD_LOGIC;
        rst   : in  STD_LOGIC;
        up    : in  STD_LOGIC;
        down  : in  STD_LOGIC;
        value : out STD_LOGIC_VECTOR(2 downto 0)
    );
end fsm;

architecture behavior of fsm is
type state_t is (S0, S1, S2, S3, S4, S5);
signal state : state_t := S0;

begin

    THREAD : process (clk)
    begin
        if (rst = '1') then
            state <= S0;
            value <= "000";
        elsif (rising_edge(clk)) then
            case state is
                when S0 => 
                    if (up = '1') then
                        value <= "001";
                        state <= S1;
                    end if;
                    if (down = '1') then
                        value <= "101";
                        state <= S5;
                    end if;
                when S1 => 
                    if (up = '1') then
                        value <= "010";
                        state <= S2;
                    end if;
                    if (down = '1') then
                        value <= "000";
                        state <= S0;
                    end if;
                when S2 => 
                    if (up = '1') then
                        value <= "011";
                        state <= S3;
                    end if;
                    if (down = '1') then
                        value <= "001";
                        state <= S1;
                    end if;
                when S3 => 
                    if (up = '1') then
                        value <= "100";
                        state <= S4;
                    end if;
                    if (down = '1') then
                        value <= "010";
                        state <= S2;
                    end if;
                when S4 => 
                    if (up = '1') then
                        value <= "101";
                        state <= S5;
                    end if;
                    if (down = '1') then
                        value <= "011";
                        state <= S3;
                    end if;
                when S5 => 
                    if (up = '1') then
                        value <= "000";
                        state <= S0;
                    end if;
                    if (down = '1') then
                        value <= "100";
                        state <= S4;
                    end if;
            end case;
        end if;
   end process;
end behavior;
```

I tried to extract the FSM as kiss, as discussed here:

using following command

```yosys '-mghdl' -p 'ghdl fsm.vhd -e fsm; proc; opt -nodffe -nosdff; fsm -nomap; fsm_export -o fsm.kiss2'```

### Expected Behavior

There should be a kiss output

### Actual Behavior

There is no kiss output

nakengelhardt, 2023-07-05:
The `fsm_extract ` pass heuristically extracts some FSMs that will benefit from recoding/optimization from the `fsm_recode` and `fsm_opt` passes. Not all kinds of FSMs can be modified in this way. This FSM has two properties that make it unsuitable for this, which the output informs you of:
```
Not marking fsm.state as FSM state register:
    Register has an initialization value.
    Circuit seems to be self-resetting.
```

closed 2023-07-05 by nakengelhardt

myzinsky, 2023-07-05:
So for me it would not be so important to optimize the FSM, rather I want to extract and plot it. Would it still be possible to plot it?

nakengelhardt, 2023-07-05:
No. The only part that can be overcome is the self-resetting part (there is the option `-ignore-self-reset` to `fsm_detect`). The other conditions are necessary for `fsm_extract` to work; it will likely crash if you disregard it.

myzinsky, 2023-07-05:
I just wonder why "Register has an initialization value." leads to the fact that this shouldn't be FSM register. Could you elaborate this a bit? I mean the state machine has to start somehow with a well defined state?

myzinsky, 2023-07-05:
I was able to extract it by changing the VHDL code a bit... however still wonder about the above question

whitequark, 2023-07-05:
@nakengelhardt I concur with @myzinsky; this was one of my earliest issue with Yosys and in all these years I never actually found out why `fsm_extract` cannot work with registers that have an init value...