YosysHQ/yosys #3837: How to figure out the actual name of the flipflop in yosys?

prajnaputhran17, 2023-07-08:
In our design there are D flipflops and fifo registers. After synthesing in yosys , it has been renamed as 
sky130_fd_sc_hd_dfrtp_2 _14258 (
    .CLK(CLK),
    .D(00157),
    .Q(mv_output[20]),
    .RESET_B(RST_N)
  );

Why the registers are not having actual names as fed in RTL?
How to figure the actual names of the flipflop in yosys?

Ravenslofty, 2023-07-08:
Does your script call `dress` or `autoname`?

prajnaputhran17, 2023-07-08:
> Does your script call `recover_names` or `autoname`?
I am using yosys in openlane and I am running with default .tcl files. I didn't find any of those commands that you have mentioned in my yosys script.

jix, 2023-07-08:
I'm not familiar with OpenLane myself, but according to this PR opt-in support (using the SYNTH_AUTONAME variable) for autoname was merged earlier this year: https://github.com/The-OpenROAD-Project/OpenLane/pull/1577

Vinayakamk, 2023-07-17:
there is no such command in openlane older version.
and this the same issue as that of this #3195 issue.