YosysHQ/yosys #3845: Improve write_verilog speed

j2kun, 2023-07-13:
### Feature Description

Based on my measurements of `write_verilog` on large designs, it writes to disk at a rate of roughly 2 MiB/s. This results in some large (~GiB) designs taking ~20-30 minutes to write to disk.

Repro: https://gist.github.com/j2kun/43b15994ddc4874c8dda26eb8fafb230

Contains `model.v` and `script.ys` which flattens the model and writes to verilog.

povik, 2023-07-19:
It's the `-noexpr` mode that feels especially slow. Anyway, with the following patch,

```
diff --git a/backends/verilog/verilog_backend.cc b/backends/verilog/verilog_backend.cc
index 3da168960..5ff191a9c 100644
--- a/backends/verilog/verilog_backend.cc
+++ b/backends/verilog/verilog_backend.cc
@@ -37,7 +37,7 @@ PRIVATE_NAMESPACE_BEGIN
 
 bool verbose, norename, noattr, attr2comment, noexpr, nodec, nohex, nostr, extmem, defparam, decimal, siminit, systemverilog, simple_lhs, noparallelcase;
 int auto_name_counter, auto_name_offset, auto_name_digits, extmem_counter;
-std::map<RTLIL::IdString, int> auto_name_map;
+dict<RTLIL::IdString, int> auto_name_map;
 std::set<RTLIL::IdString> reg_wires;
 std::string auto_prefix, extmem_prefix;
 
@@ -129,7 +129,7 @@ std::string id(RTLIL::IdString internal_id, bool may_rename = true)
                break;
        }
 
-       const pool<string> keywords = {
+       static const pool<string> keywords = {
                // IEEE 1800-2017 Annex B
                "accept_on", "alias", "always", "always_comb", "always_ff", "always_latch", "and", "assert", "assign", "assume", "automatic", "before",
                "begin", "bind", "bins", "binsof", "bit", "break", "buf", "bufif0", "bufif1", "byte", "case", "casex", "casez", "cell", "chandle",

```

the write speed was, in my testing, raised from ~2.5 MB/s to ~34 MB/s.

j2kun, 2023-07-19:
What an amazingly compact patch for such a dramatic speedup!