YosysHQ/yosys #3847: ERROR: syntax error, unexpected TOK_REAL

the-centry, 2023-07-14:
### Version

30+48

### On which OS did this happen?

Linux

### Reproduction Steps

```
module bhaskara_sin(input a,output b);
    parameter real theta = 0.0;
    parameter real sin = 0.0;
    function real sin_func (
        input real x);
        real numerator;
        real denominator; 
        real PI=3.14159265359;
        begin
            numerator = PI;
            numerator = 16 * x * (PI - x);
            denominator = (5 * PI * PI) - (4 * x * (PI - x));
            sin_func = numerator / denominator;
        end
    endfunction
    integer y;
    always @ * begin
        y = $rtoi(sin_func(theta));
    end
    assign b = y;
endmodule
```

### Expected Behavior

paser the verilog with real parameters and deal the parameters!

### Actual Behavior

ERROR: syntax error, unexpected TOK_REAL

UnsignedByte, 2024-03-21:
Recently had this same issue, looks like changing `real` to `parameter real` fixes this.

Jesaul, 2024-04-16:
Error happens on the line "function real sin_func ("

nakengelhardt, 2024-04-16:
Values of type `real` are not supported for synthesis. The only usage that is accepted is to pass them through unchanged to architecture primitives (e.g. certain PLL macros take the clock frequency as a real), but in that case yosys internally represents them as strings.