YosysHQ/yosys #3848: During synthesis, the addition of optimization passes such as opt_clean and opt_reduce after ABC optimization has resulted in errors in register assignment.

oomimo18, 2023-07-14:
### Version

 Yosys 0.30+48

### On which OS did this happen?

Linux

### Reproduction Steps

I have encountered an inconsistency issue while using Yosys. During synthesis, we did not use the default synthesis flow of Yosys. After ABC optimization, we applied additional optimization passes such as opt_clean and opt_reduce. The specific synthesis flow we followed is as follows: 
        read_verilog rtl.v
	hierarchy
	proc; opt_merge; opt_clean; opt_clean; opt_clean; opt_reduce; opt_expr;
	memory; opt_clean; opt_clean; opt_reduce; 
	techmap; opt_clean; opt_clean; opt_reduce; 
	abc; opt_demorgan; opt_clean; opt_merge; opt_clean; opt_clean; opt_clean; opt_reduce; opt_expr; opt_lut_ins; opt_expr; opt_expr; opt_clean; opt_dff; opt_lut; opt_reduce; opt_clean; opt_share; 
        write_verilog syn_yosys.v
We aimed to optimize the logical structure by adding these passes, but these modifications should not affect the code consistency. Using the provided design file (rtl.v), we performed synthesis separately with Yosys's default synthesis flow and our modified synthesis flow with additional passes. Subsequently, we conducted simulation verification using the Iverilog simulator. During simulation, we observed inconsistencies in the output signals (at 0sec). This inconsistency arose from erroneous register assignments in module5 of the design file (starting from line 50 in the attached rtl.v file). The waveform comparison chart is presented below:

[image]
Figure 1：Default Synthesis（0sec）
[image]

Figure 2：Modified Synthesis（0sec）

Please find attached the code of my program.
[rtl.txt](https://github.com/YosysHQ/yosys/files/12047822/rtl.txt)
[yosys_testbench.txt](https://github.com/YosysHQ/yosys/files/12047945/yosys_testbench.txt)

Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.

### Expected Behavior

We aimed to optimize the logical structure by adding some passes, but these modifications should not affect the code consistency. 

### Actual Behavior

However, in reality, it did exhibit inconsistent behavior during simulation verification.

povik, 2023-07-18:
Should the discrepancy be visible in the print output from running of the testbench (since it is printing out the `y` signal)? If so, I am unable to reproduce.

Once I ran the testbench with unmodified `rtl.txt`, and once on `syn_yosys.v` that resulted from your synthesis script. In both cases the first line of output is:

    000000000000000000000000000000000000010000000100000000000000000000000000000000000000000000000000000000000000

which, if I am not miscounting zeroes, is the same as you get on Figure 1.

What version of Icarus Verilog are you testing this with? Mine is 11.0. Yosys I tested with both versions 0.23 and 0.30+48.

oomimo18, 2023-07-19:
> Thank you for your response.

The Icarus Verilog version I am using is Icarus Verilog version 13.0 (devel) (s20221226-221-g272771d18).

Below are the synthesis commands used for the two synthesis processes:
1、Default synthesis process using yosys：
read_verilog rtl.v         
synth
write_verilog syn_yosys.v
2、After ABC optimization, we applied additional optimization passes such as opt_clean and opt_reduce. The specific synthesis flow we followed is as follows: 
read_verilog rtl.v
	hierarchy
	proc; opt_merge; opt_clean; opt_clean; opt_clean; opt_reduce; opt_expr;
	memory; opt_clean; opt_clean; opt_reduce; 
	techmap; opt_clean; opt_clean; opt_reduce; 
	abc; opt_demorgan; opt_clean; opt_merge; opt_clean; opt_clean; opt_clean; opt_reduce; opt_expr; opt_lut_ins; opt_expr; opt_expr; opt_clean; opt_dff; opt_lut; opt_reduce; opt_clean; opt_share; 
write_verilog  new_syn_yosys.v

We aimed to optimize the logical structure by adding these passes, but these modifications should not affect the code consistency.
However, the results show that there is inconsistency in the simulation of syn_vivado.v and new_syn_vivado.v when simulated separately using Icarus Verilog.

Default synthesis process ，the first line of output is:
000000000000000000000000000000000000010000000100000000000000000000000000000000000000000000000000000000000000
additional optimization passes in the Synthesis process， the first line of output is:
000000000000000000000000000000000000000000000100000000000000000000000000000000000000000000000000000000000000
Inconsistency occurred at bit 38. 
I think this is due to an abnormal behavior caused by some of the optimization passes, resulting in register assignment errors.
I would appreciate your assistance in identifying the root cause of the issue. Thank you very much for your help.

povik, 2023-07-19:
So I was able to reproduce after all once I paired Yosys with the right ABC version (`1de4eaf`).

povik, 2023-07-19:
This seems to happen on current master with ABC `bb64142` too

povik, 2023-07-19:
So the culprit is in `opt_share` not distinguishing between the inputs of `$_ANDNOT_` cells when considering if a mux can be moved in front. So it transforms

[image]

into

[image]

which of course is wrong.

PR #3857 (2023-07-20): opt_share: Fix input confusion with ANDNOT, ORNOT gates
Distinguish between the A, B input ports of `$_ANDNOT_`, `$_ORNOT_` gates when considering those for sharing. Unlike the input ports of the other supported single-bit gates, those are not interchangeable.

Fixes #3848.
commit f0ae046c (2023-07-20): opt_share: Fix input confusion with ANDNOT, ORNOT gates

Distinguish between the A, B input ports of `$_ANDNOT_`, `$_ORNOT_`
gates when considering those for sharing. Unlike the input ports of the
other supported single-bit gates, those are not interchangeable.

Fixes #3848.
closed 2023-07-20 by Ravenslofty