YosysHQ/yosys #3863: Liberty front end doesn't properly parse boolean expressions

Jeff-Ciesielski, 2023-07-25:
### Version

Yosys 0.25 (git sha1 e02b7f64bc7, gcc 13.1.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Attempt to parse a liberty file containing a boolean expression that utilizes a space (' ') character as a logical AND.

Example: (Apologies for not being able to share full files, foundry NDAs and all that jazz)

`next_state : "((SE SI) + (!SE D))";`

### Expected Behavior

According to the liberty users guide, this should parse as: `(SE & SI) | (!SE & D)`

[image]

### Actual Behavior

I get the following error in my yosys logs:

`Warning: Malformed liberty file - cannot find pin 'SESI+!SED' in cell '<redacted>' - skipping.`

Jeff-Ciesielski, 2023-07-25:
I see the issue, the liberty front end is just throwing away space characters which is causing the pin names to be collapsed.

liberty.cc:
```

static RTLIL::SigSpec parse_func_expr(RTLIL::Module *module, const char *expr)
{
	const char *orig_expr = expr;
	std::vector<token_t> stack;

	while (*expr)
	{
----->		if (*expr == ' ' || *expr == '\t' || *expr == '\r' || *expr == '\n' || *expr == '"') {
			expr++;
			continue;
		}

		token_t next_token(0);
		if (*expr == '(' || *expr == ')' || *expr == '\'' || *expr == '!' || *expr == '^' || *expr == '*' || *expr == '+' || *expr == '|' || *expr == '&')
			next_token = token_t(*(expr++));
		else
			next_token = token_t(0, parse_func_identifier(module, expr));

		while (parse_func_reduce(module, stack, next_token)) {}
		stack.push_back(next_token);
	}

	while (parse_func_reduce(module, stack, token_t('.'))) {}

#if 0
	for (size_t i = 0; i < stack.size(); i++)
		if (stack[i].type < 16)
			log("%3d: %d %s\n", int(i), stack[i].type, log_signal(stack[i].sig));
		else
			log("%3d: %c\n", int(i), stack[i].type);
#endif

	if (stack.size() != 1 || stack.back().type != 3)
		log_error("Parser error in function expr `%s'.\n", orig_expr);

	return stack.back().sig;
}

```

I'm going to take a crack at resolving this as this is high priority for us (fixing a ~4 million line foundry provided liberty file would likely be less effective)

povik, 2023-07-26:
It doesn't seem to be the liberty frontend proper (invoked by `read_liberty`) that spouts the message, rather it must have been the liberty parsing in `dfflibmap`. That code doesn't seem to support expressions at all (other than simple negation), and it failed to detect it's being supplied one, so it didn't print out a more helpful error message.

https://github.com/YosysHQ/yosys/blob/b04d0e09e83102e14a53bb8b8dcc8c35f63b2fbe/passes/techmap/dfflibmap.cc#L104-L113