YosysHQ/yosys #3870: “unexpected TOK_INCREMENT” when SystemVerilog source contains ++

imlibra, 2023-08-01:
### Version

“unexpected TOK_INCREMENT” when SystemVerilog source contains ++

### On which OS did this happen?

Linux

### Reproduction Steps

yosys -read -sv [hamming/hamming_dec.sv](https://github.com/dshekhalev/FEC/blob/main/rtl/hamming/hamming_parameters.svh)

### Expected Behavior

……

### Actual Behavior

-- Parsing `hamming/hamming_dec.sv' using frontend ` -sv' --

1. Executing Verilog-2005 frontend: hamming/hamming_dec.sv
Parsing SystemVerilog input from `hamming/hamming_dec.sv' to AST representation.
hamming/hamming_parameters.svh:57: ERROR: syntax error, unexpected TOK_INCREMENT

Ravenslofty, 2023-08-01:
The line in question is `generate_permutation[cDBITS + k++] = i;`, which I think is a bit more specific than just "when SystemVerilog source contains ++"; for example if Yosys didn't like ++ it'd have errored elsewhere.

zachjs, 2023-08-01:
I'm interested in adding support for assignment expressions to Yosys, but it's not something I'm currently working on. In the meantime, you could use https://github.com/zachjs/sv2v, which already supports this feature.

zachjs, 2023-09-06:
@imlibra If you have the time, please try out my patch in #3920, which adds support for assignments within expressions.

PR #3920 (2023-09-20): sv: support assignments within expressions
- Add support for assignments within expressions, e.g., `x[y++] = z;` or `x = (y *= 2) - 1;`. The logic is handled entirely within the parser by injecting statements into the current procedural block.
- Add support for pre-increment/decrement statements, which are behaviorally equivalent to post-increment/decrement statements.
- Fix non-standard attribute position used for post-increment/decrement statements.

I am not particularly attached to this implementation. I had initially anticipated adding a new AST node type and elaborating them in `simplify [... cut]
closed 2023-09-20 by povik