YosysHQ/yosys #3872: Warnings for missing backticks

thesourcerer8, 2023-08-02:
### Feature Description

Please add warnings for missing backticks for the usage of defines. I wrote a little tool which scans a whole codebase and reports all such warnings, but I think it would be very valuable if yosys did that when parsing verilog too.

mithro, 2023-08-02:
@thesourcerer8 - BTW This might be a sensible feature request for [Surelog](https://github.com/chipsalliance/Surelog/issues) (SystemVerilog frontend with [Yosys plugin](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin)) and/or [Verible](https://github.com/chipsalliance/verible) (a Verilog / SystemVerilog linter and auto formatter) too.

thesourcerer8, 2023-08-02:
I have published my tool in the mean time: https://github.com/thesourcerer8/nand_controller/blob/main/Verilog/searchdefines.pl