YosysHQ/yosys #3881: compile fails on abc/src/misc/extra/extraUtilUtil.c: "ISO C++17 does not allow 'register' storage class specifier"

salfter, 2023-08-07:
### Version

0.32

### On which OS did this happen?

Linux

### Reproduction Steps

I maintain a Gentoo Linux ebuild for Yosys and was updating to the latest version.  Compilation fails during the ABC build:

```
[ 94%] ABC: `` Compiling: /src/misc/mvc/mvcCover.c
src/misc/extra/extraUtilUtil.c:100:5: error: ISO C++17
      does not allow 'register' storage class specifier [-Wregister]
    register int c;
    ^~~~~~~~~
src/misc/extra/extraUtilUtil.c:101:5: error: ISO C++17
      does not allow 'register' storage class specifier [-Wregister]
    register const char *place;
    ^~~~~~~~~
2 errors generated.
make[1]: *** [Makefile:177: src/misc/extra/extraUtilUtil.o] Error 1
```

This error also prevents older versions of Yosys from building.  I brought up an issue with the ABC project and got it fixed in https://github.com/berkeley-abc/abc/commit/927b60b7a02e8709768fdaa1301dff2b883dd0cf.  I know that Yosys pulls ABC from its own fork, but you might want to consider rebasing against upstream.  

### Expected Behavior

Yosys builds properly with current versions of the toolchain.

### Actual Behavior

It doesn't.

whitequark, 2023-08-07:
Related/caused by https://github.com/YosysHQ/yosys/issues/2011.

salfter, 2024-02-26:
Got rid of the patch I had been using...looks like it builds fine without it now.  Thanks!