YosysHQ/yosys #3895: Inconsistency Issue with Continuous Assignment Error after FSM Optimization using opt_dff and Other Passes

oomimo18, 2023-08-16:
### Version

 Yosys 0.30+48

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,

I have come across an inconsistency problem during the synthesis process while using Yosys:
Version: Icarus Verilog version 13.0, Yosys 0.30+48

In our synthesis flow, we have deviated from using Yosys's default synthesis process and have incorporated additional optimization steps aimed at enhancing the performance, area efficiency, and other key metrics of our design. However, these changes should not affect the consistency of the code.

The two synthesis processes are as follows:

1.Using Yosys's default synthesis process:
read_verilog rtl.v         
synth
write_verilog syn_yosys.v

2.Customized synthesis process:       
	read_verilog rtl.v         
	prep -ifx; 
	hierarchy; 
	flatten; 
	proc -norom;
	opt_reduce; opt_clean;  
	memory_share; 
	opt;
	fsm_expand; opt_expr -fine; opt_clean -purge; opt_share; opt_expr -keepdc; opt_dff -nodffe; opt_expr; 
	abc;   
	write_verilog syn_yosys.vv
After obtaining synthesis files from both of the above processes, I performed simulation verification using the Iverilog simulator. However, I observed inconsistencies in the output signals during simulation, as highlighted in the red-boxed section in the image below.
Default synthesis process，the first second of output is:
[image]
Modified Synthesis， the second line of output is:
[image]
Please find attached the code of my program.
Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.
[yosys_example.zip](https://github.com/YosysHQ/yosys/files/12358898/yosys_example.zip)

### Expected Behavior

Consistent simulation results

### Actual Behavior

Inconsistent simulation results

oomimo18, 2023-09-07:
@kivikakk @whitequark @povik @esden @pda @thoughtpolice 
Hello everyone, the simulation results have troubled me for a long time. Can you help me to find out the reason of this problem, thank you very much.

povik, 2023-09-07:
Hello @oomimo18, thanks for writing down a reasonable bug report. This being a community project, support is only provided to the extent that individual community members are interested in a particular issue. I point that out because tagging a number of unrelated people like you have just done can be seen as rude. As an option there's YosysHQ and other parties you could contract out to, to receive support and resolution of issues in a timely manner on a commercial basis.

That being said, I happen to have looked at your issue earlier. It seemed like inputting `rtl.v` directly into iverilog leads to yet another simulation behavior distinct from the two other cases you describe. After some inspection I decided there's too much going on with your issue, and gave up. Not that I don't want to see the underlying problems fixed but I can't invest the time right now.

povik, 2023-09-07:
Just to clarify, I sort of understand why you would tag me, but tagging five other people crosses a line. Anyway, you seem to understand, so let's leave it be. I will get back to you if I find anything on your issue.