YosysHQ/yosys #3898: i386 (only) test failure: Error opening grom.fst as FST file

DanielG, 2023-08-22:
Hi,

I'm seeing the `test/sat/grom.ys` test fail on Debian i386 (but not on amd64 or any other architecture). I tried to bisect the issue in the YosysHQ/yosys repo, but even as far back as yosys-0.23 I'm seeing the test fail now altough it must have passed just fine as part of the 0.23 Debian package builds see buildd history below. 

Repro, in a Debian i386 chroot:

```
$ echo CONFIG:=gcc > Makefile.conf
$ make -j32
$ ( cd tests/sat/; ../../yosys grom.ys )
[...]
7. Executing SIM pass (simulate the circuit).
ERROR: Error opening 'grom.fst' as FST file
```

Full grom.log in yosys repo on top of 6405bbab1e8afc587febf41339aaf8b514c8f202 (Fri Aug 18 00:14:07 2023 +0000): https://gist.github.com/DanielG/084cf786c57afc4050dd5f874bee7869
Debian i386 buildd log: https://buildd.debian.org/status/fetch.php?pkg=yosys&arch=i386&ver=0.30-2&stamp=1692708361&raw=0
Debian i386 buildd history, showing that 0.23 should work: https://buildd.debian.org/status/logs.php?pkg=yosys&arch=i386 

A quick debug session shows this break being triggered:
```
                        if (vu.d != FST_DOUBLE_ENDTEST) {
                            break; /* either corrupt file or wrong architecture (offset +33 also functions as matchword)
                                    */
                        }
```

but I haven't had time to dig further into it yet.

--Daniel

DanielG, 2023-08-22:
Here's a gdb session log with commentary.

```
Breakpoint 7, fstReaderInit (xc=<optimized out>) at libs/fst/fstapi.cc:4337
4337	                    xc->double_endian_match = (dcheck == FST_DOUBLE_ENDTEST);
(gdb) p dcheck 
$15 = 2.7182818284590451
N:    2.7182818284590452354  <- #define FST_DOUBLE_ENDTEST (2.7182818284590452354)
N: looks like a precision issue, my guess is FST_DOUBLE_ENDTEST is missing the "L" "doubLe" literal postfix. If so, how this ever worked is a mystery.
(gdb) n
4338	                    if (!xc->double_endian_match) {
4349	                            vu.rvs_buf[rvs_idx] = dcheck_alias[7 - rvs_idx];
[...]
4348	                        for (rvs_idx = 0; rvs_idx < 8; rvs_idx++) {
4351	                        if (vu.d != FST_DOUBLE_ENDTEST) {
N: The "break" doesn't show up in gdb but it's being hit here.
N: return is (0) here.
fstReaderOpen (nam=0xffffd240 "grom.fst") at libs/fst/fstapi.cc:4530
4530	        if ((rc) && (xc->vc_section_count) && (xc->maxhandle) &&
```

jix, 2023-08-22:
Does compiling the code with `-ffloat-store` fix this issue (or alternatively `-mfpmath=sse` which should be faster but that's not compatible with sufficiently ancient 32-bit x86 cores and floating point performance isn't critical for yosys anyway)? What you describe looks like it could be an issue with x87 excess precision, although I haven't looked at the code you mentioned in context or tried reproducing it to confirm...

DanielG, 2023-08-22:
I tried both as well as `-fexcess-precision=standard`, was also suggested in `#debian-science`, but none changed the behavior.

Does the excess precision also apply to equality checks? As that might explain why the check is broken but not why twiddling the compiler options doesn't fix it then. My theory is the constant being compared against (FST_DOUBLE_ENDTEST) is stored at higher precision than the double being read from the file which got truncated to 64bit already.

```
#define FST_DOUBLE_ENDTEST (2.7182818284590452354)
```

DanielG, 2023-08-22:
Yup, this unholy patch "fixes" the error:

```
diff --git a/libs/fst/fstapi.cc b/libs/fst/fstapi.cc
index da0d959a0..5f0d83e06 100644
--- a/libs/fst/fstapi.cc
+++ b/libs/fst/fstapi.cc
@@ -4326,6 +4326,8 @@ int fstReaderInit(struct fstReaderContext *xc)
             if (sectype == FST_BL_HDR) {
                 if (!hdr_seen) {
                     int ch;
+		    double ref = FST_DOUBLE_ENDTEST, refbuf;
+		    memcpy(&refbuf, &ref, sizeof(ref));
                     double dcheck;
 
                     xc->start_time = fstReaderUint64(xc->f);
@@ -4334,7 +4336,7 @@ int fstReaderInit(struct fstReaderContext *xc)
                     hdr_incomplete = (xc->start_time == 0) && (xc->end_time == 0);
 
                     fstFread(&dcheck, 8, 1, xc->f);
-                    xc->double_endian_match = (dcheck == FST_DOUBLE_ENDTEST);
+                    xc->double_endian_match = (dcheck == ref);
                     if (!xc->double_endian_match) {
                         union
                         {
-- 
```

my theory being that by forcing the roundtrip through memory the internal 80-bit representation is forced to be truncated to a normal 64bit double. Still puzzled why the gcc option didn't take care of this.

jix, 2023-08-22:
It might also be worth checking what Debian does for gtkwave from which the fst code is taken.

DanielG, 2023-08-22:
Nothing, I see no patches or CXXFLAGS changes. and the code at src/helpers/fst/fstapi.c smells pretty much the same. Only difference I can detect is .c instead of .cc as in yosys, perhaps this is a C vs C++ ABI behavioural issue.

jix, 2023-08-22:
This seems relevant https://gcc.gnu.org/bugzilla/show_bug.cgi?id=323#c225 especially the part right below "The existing testcases tweaks" is about checks like this, so it seems to be expected behavior for gcc since this patch got merged.

DanielG, 2023-08-22:
Right, that gave me an idea. Casting both sides of the comparison to double seems to fix the issue too:

```
       xc->double_endian_match = ((double)dcheck == (double)FST_DOUBLE_ENDTEST);
```

That's probably a more stomachable patch.

--Daniel

DanielG, 2025-03-18:
@KrystalDelusion We believe 2edb9397c317cb3ef3e4d29741069e0cd900ea reintroduced this issue. Could you reopen? GH won't let me.

DanielG, 2025-03-24:
TBH I hate GH Actions with a passion so I don't really want to touch it :D. I could offer a yosys mirror on Debian's Gitlab instance that does i386 builds, but that seems too much hassle to be worth it.

It's fine as-is really this is what we have autpkgtest for :-).

My only recommendation would be to upstream the patch into gtkwave proper. I really just missed that this code came from gtkwave last time round. If you don't have time I can do that, just LMK.

I may get around to adding the patch to the gtkwave Debian package since I suspect it's affected too it just doesn't exercise that code in the build infra.

--Daniel

KrystalDelusion, 2025-03-24:
> It's fine as-is

Sweet as then :)

> My only recommendation would be to upstream the patch into gtkwave proper. I really just missed that this code came from gtkwave last time round. If you don't have time I can do that, just LMK.

That would be good if you can, since as mentioned we don't have a 32bit test setup to reproduce this ourselves.

commit e017f660 (2023-08-27): Fix i386 FP excess-precision issue in fstapi (Fixes: #3898)

Likely related to https://gcc.gnu.org/bugzilla/show_bug.cgi?id=323#c225

Thanks to @jix for digging this up
PR #4953 (2025-03-24): A few fixes, mostly local docs builds
_What are the reasons/motivation for this change?_
- Close #3898 (again).
- Local html builds included `READTHEDOCS` et al, making a weird and unformatted version selector element in the sidebar.
- Close #4777.

_Explain how this is achieved._
- Reapply #3904, now with patch file.
- Gate `READTHEDOCS` on environment variable.
- Drop svg badges.

_If applicable, please suggest to reviewers how they can test the change._
commit 013206de (2025-03-24): Fix #3898 (again)
closed 2025-03-24 by KrystalDelusion