YosysHQ/yosys #3905: latexpdf docs build fails due to emoji character

DanielG, 2023-08-27:
Hi,

I'm seeing the following fatal error when running `make docs DOC_TARGET=latexpdf`: 

```
! LaTeX Error: Unicode character 👇 (U+1F447)
               not set up for use with LaTeX.
```

Removing the entirely unnecessary emojis fixes it.

--Daniel

DanielG, 2023-08-27:
@KrystalDelusion your #3783 seems to be at fault here.

KrystalDelusion, 2023-08-27:
Thanks for catching that, I'm curious how Read The Docs gets around this since it seems to build the pdf fine (albeit without the emoji)

Ravenslofty, 2023-08-30:
I somewhat disagree that the emoji (which through a [draft blog post](https://gist.github.com/Ravenslofty/0e757ae8262b671b23f2cfdbe81eec9a) I was writing got adopted by YosysHQ and eventually made a documentation entry) are "entirely unnecessary"; it's a reasonable if stylistic way of getting attention to the change in question (I adopted it from [fasterthanlime](https://fasterthanli.me/)).

commit 70c47690 (2023-08-27): Updating todo text and assorted fixes

Fix #3905 by removing emoji (and move the comment into the if block for less ambiguity).
Adds `latexmk` to README.

Note that latexpdf doesn't seem to like `cmd:ref` links, possibly because the
reference location is inside a latex comment block, but I was under the
impression that there was a reference location in there previously which was
working fine.  May be related to how the `cmd:def` block expands (or doesn't as
the case may be).
closed 2024-03-18 by nakengelhardt