YosysHQ/yosys #3911: Allow disabling NodeFS dependency for Emscripten

DismissedGuy, 2023-08-31:
### Feature Description

We currently compile both Yosys and Nextpnr using Emscripten to be able to use the tools in both Web and Node.js worker environments. This means that, instead of using nodefs, we communicate the tool inputs/outputs by posting messages between the host and the worker process. However, Yosys emits JavaScript code ([link](https://github.com/YosysHQ/yosys/blob/b739213d9f00b409ef7b9a9e1834d659ea8008d9/kernel/driver.cc#L101-L109)) that contains a hard dependency on nodefs, which is actually causing us issues when running under the Node.js worker. We currently get around this limitation by stripping these mount calls from our final bundle, however this is of course not ideal.

I would like to propose a compilation flag which, when enabled, would simply not emit those lines of code. This would allow users to take care of the filesystem logic themselves.

Note that these same lines of code also occur in IceStorm ([search result](https://github.com/search?type=code&auto_enroll=true&q=NODEFS+owner%3AYosysHQ)); while we haven't run into this issue with Nextpnr yet, it would be nice if those lines could be disabled in that tool as well.

whitequark, 2023-08-31:
One of the options you have is to use the [YoWASP](https://yowasp.org) binaries which are pure-WASM and allow you to use any FS implementation you want.

DismissedGuy, 2023-08-31:
I *believe* we've looked into that already, but I'm not entirely sure if that's true or why it was decided against using it. @DanielleHuisman maybe you could shine some light on this?

DanielleHuisman, 2023-09-01:
We are still trying to find a WASI implementation that works with the YoWASP binaries and suites our needs. So yes, in the future we hope to use those binaries instead of the Emscripten ones.

whitequark, 2023-09-01:
> We are still trying to find a WASI implementation that works with the YoWASP binaries and suits our needs. So yes, in the future we hope to use those binaries instead of the Emscripten ones.

I have this implementation locally and I have had it tested with yowasp-yosys. If this is something you're interested in seeing released I'm happy to have you contract me to clean it up and publish.

malmeloo, 2024-04-17:
No longer necessary now that we're using yowasp 🙂

closed 2024-04-17 by malmeloo