YosysHQ/yosys #3917: `$print` cells converted from `$display` tasks don't sample signals according to Verilog semantics

povik, 2023-09-05:
### Version

Yosys 0.32+76 (git sha1 73cb4977b2e, clang++ 11.1.0 -fPIC -Os) with patch for print support in sim command

### On which OS did this happen?

macOS

### Reproduction Steps

Take the following script

```
read_verilog <<EOF
module top(clk);
input wire clk;

reg [7:0] counter = 0;
always @(posedge clk) begin
    counter = counter + 1;
end

reg [7:0] save;
always @(posedge clk) begin
    save = counter;
    $display("counter %x save %x", counter, save);
end
endmodule
EOF
prep
sim -q -clock clk
```

### Expected Behavior

`counter` and `save` have the same value as far as the print is concerned, i.e. the following gets printed

```
counter 00 save 00
counter 01 save 01
counter 02 save 02
counter 03 save 03
counter 04 save 04
counter 05 save 05
...
```

This is what happens with iverilog.

### Actual Behavior

Log output

```
counter 01 save 00
counter 02 save 01
counter 03 save 02
counter 04 save 03
counter 05 save 04
counter 06 save 05
...
```

Design after `prep`

[image]

whitequark, 2023-09-05:
As the person who did most of the `$display` work I'm tempted to say this is out of scope. Yosys is a synthesizer, not a Verilog interpreter. There are other places where `sim` would have a mistmatch with `iverilog`, surely.

povik, 2023-09-05:
Sure, it's not, but the synthesizable subset should agree with an interpreter, or this is not `$display`. I don't think implementing this properly would be as bad if we inject assignments to dummy signals at the place the `$display` statement is found, then leave the general process machinery deal with those assignments.

PR #3918 (2023-09-11): ast: Substitute rvalues when parsing out print arguments
Apply the local substitutions stemming from process context when parsing out format arguments to `$display` or other statements.

This is to fix #3917.