YosysHQ/yosys #3942: "write_json" sometimes preserves the leading "\" of an escaped identifier, sometimes not.

YikeZhou, 2023-09-19:
### Version

Yosys 0.33+53 (git sha1 e2b613355, clang 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Verilog file `top.v`:
```verilog
module top;
wire \0a , \a ;
endmodule
```
Yosys script:
```
read_verilog top.v
hierarchy -check -top top
write_json top.json
```

### Expected Behavior

All identifiers either have the leading "\\", or none of them have it.

### Actual Behavior

`top.json` generated by Yosys:
```json
{
  "creator": "Yosys 0.33+53 (git sha1 e2b613355, clang 14.0.0-1ubuntu1.1 -fPIC -Os)",
  "modules": {
    "top": {
      "attributes": {
        "top": "00000000000000000000000000000001",
        "src": "top.v:1.1-3.10"
      },
      "ports": {
      },
      "cells": {
      },
      "netnames": {
        "\\0a": {
          "hide_name": 0,
          "bits": [ 2 ],
          "attributes": {
            "src": "top.v:2.6-2.9"
          }
        },
        "a": {
          "hide_name": 0,
          "bits": [ 3 ],
          "attributes": {
            "src": "top.v:2.12-2.14"
          }
        }
      }
    }
  }
}
```

One of the netnames is "\\\\0a", while the other is "a".

nakengelhardt, 2023-09-19:
RTLIL does not distinguish escaped or unescaped identifiers, so `a` and `\a` are indistinguishable internally in yosys. Since both would be designating the same wire in verilog, write_json only escapes identifiers that wouldn't be valid unescaped.

What problems is this causing for you? And what would the expected output name be if the code uses both `\a` and `a`?

YikeZhou, 2023-09-19:
I must apologize for providing such an inappropriate example. Please check this one:
```verilog
module top;
wire \a[ ;
endmodule
```
`a[` is not a simple identifier, but write_json doesn't escape it either.
```
{
  "creator": "Yosys 0.33+53 (git sha1 e2b613355, clang 14.0.0-1ubuntu1.1 -fPIC -Os)",
  "modules": {
    "top": {
      "attributes": {
        "top": "00000000000000000000000000000001",
        "src": "top.v:1.1-3.10"
      },
      "ports": {
      },
      "cells": {
      },
      "netnames": {
        "a[": {
          "hide_name": 0,
          "bits": [ 2 ],
          "attributes": {
            "src": "top.v:2.6-2.9"
          }
        }
      }
    }
  }
}
```

nakengelhardt, 2023-09-19:
You're right, my bad; there are additional escaping rules used in `write_verilog` but `write_json` doesn't have to follow verilog rules and uses the result of the IDString `unescape_id` function directly. https://github.com/YosysHQ/yosys/blob/8fb807cd242207e40e9040cfa004d02205f12a64/kernel/rtlil.h#L405-L415

As far as I can tell from how it's used, this is considered the "display name" of a design element after it's been processed by yosys. I'm not sure what went into these particular rules for when to strip the leading `\`.