YosysHQ/yosys #3944: SEGFAULT when reading attached Verilog and running hierarchy

oharboe, 2023-09-19:
### Version

Yosys 0.33+34 (git sha1 b84ed5d3a, gcc 12.3.0-1ubuntu1~23.04 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Unzip [forward.zip](https://github.com/YosysHQ/yosys/files/12659151/forward.zip)

```
$ yosys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33+34 (git sha1 b84ed5d3a, gcc 12.3.0-1ubuntu1~23.04 -fPIC -Os)

yosys> read_verilog forward.v 
1. Executing Verilog-2005 frontend: forward.v
Parsing Verilog input from `forward.v' to AST representation.
Generating RTLIL representation for module `\RegisterFile'.
Generating RTLIL representation for module `\ForwardingMux'.
Generating RTLIL representation for module `\RegisterFileWithForwarding'.
Successfully finished Verilog frontend.

yosys> hierarchy 

2. Executing HIERARCHY pass (managing design hierarchy).
Segmentation fault (core dumped)
```

### Expected Behavior

No crash

### Actual Behavior

Crash

gatecat, 2023-09-19:
This seems to be a duplicate of https://github.com/YosysHQ/yosys/issues/1689 - `RegisterFile` is trying to instantiate itself, and this currently crashes yosys.

oharboe, 2023-09-19:
Already reported in #1689

closed 2023-09-19 by oharboe