YosysHQ/yosys #3948: Multibit `reg` crashes Yosys without an error

gussmith23, 2023-09-21:
### Version

Yosys 0.31+49 (git sha1 e0ba07aed, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

Try to `read_verilog -sv` on the following file:

```verilog
module m(input [1:0] in, output [1:0] out);
  buf buf0[1:0] (out, in);
endmodule
```

### Expected Behavior

No crash.

### Actual Behavior

```

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.31+49 (git sha1 e0ba07aed, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)

yosys> read_verilog -sv tmp.v 
1. Executing Verilog-2005 frontend: tmp.v
Parsing SystemVerilog input from `tmp.v' to AST representation.
Generating RTLIL representation for module `\m'.
libc++abi: terminating with uncaught exception of type std::out_of_range: vector
[1]    52823 abort      yosys
```

whitequark, 2023-09-21:
Have you been fuzzing Yosys?

gussmith23, 2023-09-21:
no, but I do expose it to a lot of Verilog from different sources! so
kinda!