YosysHQ/yosys #3950: In cxxrtl case item expressions are compared in an unexpected order

YikeZhou, 2023-09-22:
### Version

Yosys 0.33+65 (git sha1 934c82254, clang 14.0.0-1ubuntu1.1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Verilog file `top.v`:
```Verilog
module top (
    input clk,
    input [1:0] select,
    output reg [1:0] result
);
always @(posedge clk) begin
    case (select)
        2'b00: result <= 2'b11;
        2'b01: result <= 2'b10;
        2'b10: result <= 2'b01;
        default: result <= 2'b00;
    endcase
end
endmodule
```
Command:
```
yosys -p "read_verilog -formal top.v; hierarchy -check -top top; write_cxxrtl top.cpp"
```

### Expected Behavior

According to section 9.5 of Verilog Standard:
> The case expression given in parentheses shall be evaluated exactly once and before any of the case item expressions. The case item expressions shall be evaluated and compared **in the exact order in which they are given**. If there is a default case item, it is ignored during this linear search. During the linear search, if one of the case item expressions matches the case expression given in parentheses, then the statement associated with that case item shall be executed, and the linear search shall terminate. 

### Actual Behavior

By checking `p_top::eval()` in `top.cpp`, I found it first evaluates the third item in the case expression.
```cpp
p_result.next =
	(eq_uu<1>(p_select, value<2>{0x2u})
		? value<2>{0x1u}
		: (eq_uu<1>(p_select, value<2>{0x1u}) ? value<2>{0x2u} : (logic_not<1>(p_select) ? value<2>{0x3u} : value<2>{0u})));
```

---

In the situation above, the comparison order won't lead to incorrect results in the simulation. However, the order does matter when there are overlaps between case items.

Here is another example I found in Verilog Standard:
[image]

Therefore, it may be necessary to address this issue.

whitequark, 2023-09-22:
CXXRTL does not follow the Verilog evaluation rules, but the synthesis evaluation rules. In synthesis the evaluation order does not matter. Therefore in my view this report is invalid.

povik, 2023-09-22:
As far as I can see there's not an issue. What you are looking at in the CXXRTL generated source *does* functionally corresond to the Verilog input, and is the result of the `proc` pass (invoked implicitly by `write_cxxrtl`) implementing the case construction with a tree of muxes. You are wrong in extrapolating the order of the muxes to imply that putting in the example from the Verilog standard would lead to incorrect results.

Try running `write_cxxrtl -noproc`, which will reconstruct the original case statement more closely.

YikeZhou, 2023-09-22:
Thank you for your prompt replies.

So, this is a mismatch between pre-synthesis and post-synthesis simulations, and what cxxrtl does is a post-synthesis simulation. And the errors I came across (when using `casex`) are actually design problems.
🤔 This does resolve my confusion. I will close this issue then!

closed 2023-09-22 by YikeZhou