YosysHQ/yosys #3960: CXXRTL output doesn't compile with zero-sized ports on modules

povik, 2023-09-28:
### Version

Yosys 0.32 (git sha1 fbab08acf14, clang++ 11.1.0 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

Compiling a design with an instance of a module with a zero-sized port (as an edge case in the module's specialization)

### Expected Behavior

Compilation goes through

### Actual Behavior

Compilation of the CXXRTL output stumbles on
```
/include/backends/cxxrtl/cxxrtl.h:1095:3: error: static_assert failed due to requirement 'sizeof (item) == value<0>::chunks * sizeof(unsigned int)' "value<Bits> is not compatible with C layout"
                static_assert(sizeof(item) == value<Bits>::chunks * sizeof(chunk_t),
                ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I assume this is due to empty structs not having zero size in C++.

udif, 2023-09-28:
AFAIK, zero sized ports are not legal verilog or systemverilog code.
The typical [WIDTH-1:0] construct yields an error when WIDTH==0 on any tool I'm aware of.

povik, 2023-09-28:
> AFAIK, zero sized ports are not legal verilog or systemverilog code. The typical [WIDTH-1:0] construct yields an error when WIDTH==0 on any tool I'm aware of.

Now that I think of it, strictly speaking shouldn't the Verilog construct be interpreted to declare a two-bit vector with reversed indexing when `WIDTH=0`? I think that's how Yosys does interpret it, but if I have a CXXRTL blackbox with explicit `(* cxxrtl_width = "WIDTH" *)` annotation on the port, then CXXRTL unsurprisingly takes the `WIDTH=0` case to imply a zero-sized port. That's the behavior I want but if this wouldn't work outside of CXXRTL the same way then I guess the value in supporting this is dubious. That is, there's no companion synthesis flow where I would supply a Verilog implementation of what's otherwise the CXXRTL-blackboxed module, and the specialization would match what happens on the CXXRTL side.

I will close this for the time being.

closed 2023-09-28 by povik

udif, 2023-09-28:
> Now that I think of it, strictly speaking shouldn't the Verilog construct be interpreted to declare a two-bit vector with reversed indexing when WIDTH=0?

It seems you were right!

Consider the following test:
```
module x (
  input [6:5]w1,
  input [5:6]w2,
  input [-5:-6]w3,
  input [-6:-5]w4,
  input [0:-1]w5,
  input [-1:0]w6
);
  generate
    $info("w1=%1d w2=%1d w3=%1d w4=%1d w5=%1d w6=%1d", $bits(w1), $bits(w2), $bits(w3), $bits(w4), $bits(w5), $bits(w6));
  endgenerate
endmodule
```

I've tested 4 tools:
slang, Verilator and 2 commercial tools, and they all produce `w1=2 w2=2 w3=2 w4=2 w5=2 w6=2` .

whitequark, 2023-09-29:
> but if I have a CXXRTL blackbox with explicit `(* cxxrtl_width = "WIDTH" *)` annotation on the port, then CXXRTL unsurprisingly takes the `WIDTH=0` case to imply a zero-sized port.

We can special-case that to just remove the port entirely, maybe?