YosysHQ/yosys #3976: segmentation fault

jalcim, 2023-10-03:
### Version

Yosys 0.33 (git sha1 2584903a0, gcc 12.2.0-14 -Og -O3 -DNDEBUG -fPIC)

### On which OS did this happen?

Linux

### Reproduction Steps

crashtest.sh
```tcl
yosys -import

set WAY  2
set WIRE 32

set NAME "Dlatch"

read_verilog $NAME.v
chparam -set "WAY" $WAY $NAME
chparam -set "WIRE" $WIRE $NAME

#segfault inside
hierarchy -check -top $NAME
check

write_verilog out.v
write_spice out.sp
write_rtlil out.rtlil
```

Dlatch.v
 ```verilog
 `ifndef __DLATCH__
 `define __DLATCH__

 `include "parallel_Dlatch.v"
 `include "serial_Dlatch.v"

module Dlatch (D, clk, Q, QN);
   parameter WAY = 1;
   parameter WIRE = 1;

   input [WAY -1: 0] clk;
   input [(WAY * WIRE)-1 : 0] D;
   output [(WAY * WIRE)-1:0]  Q, QN;

   wire [4:0] line;

   if (WAY > 1)
        parallel_Dlatch #(.WAY(WAY), .WIRE(WIRE)) parallel_Dlatch_inst(D, clk, Q, QN);
   else if (WIRE > 1)
        serial_Dlatch #(.WIRE(WIRE)) inst0(a, clk, s1, s2);
   else
     begin
        not not0(line[0], D);

        and and0(line[1], line[0], clk);
        and and1(line[2], D      , clk);

        nor nor0(line[3], line[1], line[4]);
        nor nor1(line[4], line[2], line[3]);

        assign Q  = line[3];
        assign QN = line[4];
     end

endmodule

`endif
 ```
 
parallel_Dlatch.v
 ```verilog
 `ifndef __PARALLEL_DLATCH__
 `define __PARALLEL_DLATCH__

`include "Dlatch.v"
`include "serial_Dlatch.v"

module parallel_Dlatch(D, clk, Q, QN);
   parameter WAY = 3;
   parameter WIRE = 8;

   input [WAY-1:0] clk;
   input [(WAY*WIRE)-1 : 0] D;
   output [(WAY*WIRE)-1 : 0] Q, QN;

   if (WIRE == 1)
     Dlatch Dlatch0(D[0], clk[0], Q[0], QN[0]);
   else if (WIRE > 1)
     serial_Dlatch #(.WIRE(WIRE)) Dlatch1(D[WIRE-1:0], clk[0], Q[WIRE-1:0], QN[WIRE-1:0]);

   if (WAY > 1)
     parallel_Dlatch #(.WAY(WAY-1), .WIRE(WIRE)) parallel_Dlatch0(D[WAY*WIRE-1 : WIRE],
                                                                  clk[WAY-1:1],
                                                                  Q[WAY*WIRE-1 : WIRE],
                                                                  QN[WAY*WIRE-1 : WIRE]);
endmodule

`endif

 ```
 
serial_Dlatch.v
 ```verilog
 
 File Edit Options Buffers Tools Statements Verilog Help                                                                                                                                           
`ifndef __SERIAL_DLATCH__
 `define __SERIAL_DLATCH__

 `include "Dlatch.v"

module serial_Dlatch(D, clk, Q, QN);
   parameter WIRE = 1;

   input          clk;
   input [WIRE -1:0] D;
   output [WIRE -1:0] Q, QN;

   if (WIRE > 0)
     Dlatch latch1(D[0], clk, Q[0], QN[0]);
   if (WIRE > 1)
     serial_Dlatch #(.WIRE(WIRE-1)) recall(.D(D[WIRE-1:1]),
                                           .clk(clk),
                                           .Q(Q[WIRE-1:1]),
                                           .QN(QN[WIRE-1:1]));
endmodule

`endif
```

Note : I try tu use gdb on yosys, but without success.

Makefile.conf
```Makefile
ENABLE_DEBUG :=1
ENABLE_NDEBUG := 1
```

```shell
$ gdb yosys

Reading symbols from yosys...
(No debugging symbols found in yosys)
```

### Expected Behavior

synthesize

### Actual Behavior

segmentation fault

jalcim, 2023-10-03:
i find no solution for launch yosys with gdb

povik, 2023-10-03:
As far as I can tell this is a duplicate of #1689. In this case the recursion is from `Dlatch` instancing `parallel_Dlatch` instancing `serial_Dlatch` instancing `Dlatch` again.

This recursive chain is broken if you specify the values for the parameters in instancing of `Dlatch` from `serial_Dlatch`:
```
  if (WIRE > 0)
    Dlatch #(.WIRE(1), .WAY(1)) latch1(D[0], clk, Q[0], QN[0]);
```

closed 2023-10-03 by povik