YosysHQ/yosys #3993: ABC9 "&mfs" step gives incorrect result on big endian for objects with >5 variables

zeldin, 2023-10-08:
### Version

Yosys 0.33 (git sha1 2584903a060, gcc 12.3.1_p20230526 -O2 -mcpu=power9 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

A [designed created with LiteX](https://github.com/pottendo/RVCop64-pottendo) containing a VexRiscv-SMP core with 64-bit databus was synthesized with the same yosys version on ppc64 and ppc64le, both with and without `abc9.nomfs`

### Expected Behavior

The same (working) JSON should be generated on both ppc64 and ppc64le, also without `abc9.nomfs`.

### Actual Behavior

With `abc9.nomfs` both ppc64 and ppc64 syntesize the same JSON, from which a working bitstream can be generated.
However, without `abc9.nomfs` different JSONs are generated on ppc64 and ppc64le, and only the one geneated by ppc64le can be turned into a working bitstream.

The reason is because `giaMfs.c` makes used of something referred to as "truth tables", which are really bit vectors.  Rather than using a portable opaque implementation of a bit vector, UCB have opted to create a non-portable non-opaque implementation.

The implementation uses a vector of limbs to represent a bit vector of arbitrary but constant length.  The limb containing a particular bit with number `n` is chosen as
```
limbs[n >> log2(limb_bits)]
```
To extract the bit from the limb by moving it to the LSB (as is done for example by `Abc_TtGetBit()`), an expression like
```
limb >> (n & (limb_bits-1))
```
is used.

So far this is standard fare.  The problem arises because the code accesses the same bit vector alternatingly with a limb size of 32 bits and 64 bits.  Because of how the indexing is constructed above, this gives a consistent view on little endian, but not on big endian.  Due to the periodic way small vectors are initialized, it seems to work by happy accident for vectors with 32 or less significant bits, but breaks on larger ones.

This can be fixed in two ways:

1.  Select one limb size to be the canonical limb size.
Accesses using this limb size can pick any indexing scheme, including the above, regardless of endianness.  However, accesses using the non-canonical limb size would need to take into account that the ordering of 32 bit limbs within 64 bit limbs differs between big and little endian.  Bit-parallel operations like AND and OR can ignore this difference (except for the special case of a vector consisting only of a single 32-bit limb, although this case could be eliminated by always allocating at least 64 bits), but code doing shifts of containing literal limb values which are not 32-bit periodic would need to be updated.

or

2. Modify the indexing depending on endianness to preserve a consistent view over different limb sizes.
There is of course nothing magical about little endian that makes the switching of limb size work, it's just that the indexing scheme has been constructed to make it work on little endian specifically.  Since the limb index is `n >> log2(limb_bits)`, it means that the bit with number 0 is stored in the _first_  (index 0) limb.  In order for the indexing to be consistent over limb size, it is therefore important that the bit with number 0 is also stored in the _first_ bit of the limb.  On little endian this means the LSB, but on big endian this is the MSB.  So with this arrangement, on big endian the code to move bit n into the LSB of the limb would become
```
limb >> ((~n) & (limb_bits-1))
```
instead, while
```
limb << ((n) & (limb_bits-1))
```
would move bit n to the MSB of the limb.
With this approach code does not need to care about the limb size, but any and all code which does shifts or which contains non-palindromic literal limb values would need updating.

Regardless of which of these approaches is chosen, the lack of abstraction in the code makes updating it a daunting task.  A good first step would probably be to increase the use of the abstractions that do exist.  For example, `Extra_PrintHex()` could easily have used `Abc_TtGetHex()`, but does not.  There are also multiple instances of the same constant truth tables, like the truth table `s_Truths6` in `utilTruth.h` which exists in verbatim copies in at least `giaTruth.c` and `giaMfs.c`...

Ravenslofty, 2023-10-08:
Could you file this against [ABC](https://github.com/berkeley-abc/abc/issues) instead of Yosys? This isn't in our code.

closed 2023-10-08 by Ravenslofty

zeldin, 2023-10-08:
Yosys uses a fork of ABC, which does not have issue tracking enabled, which is why I figured it was better to report it here.  But I can take it upstreams if you like.

Ravenslofty, 2023-10-08:
It's true that we have a fork of ABC, but we follow upstream very closely.

And, as the Yosys ABC9 maintainer, let me tell you that I find the ABC code equally difficult to work with; but the effort that would be required to replace it with something cleaner and more maintainable would be immense, because ABC has lots of little tricks to improve QoR that are not documented.

whitequark, 2023-10-09:
Right, because who cares that we're outputting an incorrect netlist as long as it toggles really fast? 😇

Ravenslofty, 2023-10-09:
Who knows, perhaps something might become of [toymap](https://github.com/povik/toymap).

whitequark, 2023-10-09:
We could also compile abc to wasm, run it thru wasm2c, and it will work fine on any platform, LE or BE.

Ravenslofty, 2023-10-09:
And the code will be about as readable as before. Sounds good to me!

whitequark, 2023-10-09:
Perfect! Looking forward to this solution being merged then.