YosysHQ/yosys #4010: Synthesis optimization error, inconsistent simulation results

oomimo18, 2023-10-23:
### Version

 Yosys 0.30+48

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,

I have come across an inconsistency problem during the synthesis process while using Yosys:
Version: Icarus Verilog version 13.0, Yosys 0.30+48

In our synthesis flow, we have deviated from using Yosys's default synthesis process and have incorporated additional optimization steps aimed at enhancing the performance, area efficiency, and other key metrics of our design. However, these changes should not affect the consistency of the code.

The two synthesis processes are as follows:

1.Using Yosys's default synthesis process:
read_verilog rtl.v         
synth
write_verilog syn_yosys.v

2.Customized synthesis process:       
	read_verilog rtl.v         
	hierarchy; 
	flatten; 
	proc -norom;
	proc -noopt; 
	fsm_extract; opt_share; opt_reduce; wreduce; opt_mem_feedback; opt_dff -keepdc; opt_dff -sat; opt_reduce -full; opt_clean; opt_expr -mux_bool; opt_dff -nodffe; opt_expr -keepdc; opt_dff -nodffe; opt_reduce -fine; opt_expr; opt_expr -fine;
	techmap;
	abc;   
	write_verilog syn_yosys.v
After obtaining synthesis files from both of the above processes, I performed simulation verification using the Iverilog simulator. However, I observed inconsistencies in the output signals during simulation, as highlighted in the red-boxed section in the image below.  Displaying "00000" and "10111" respectively. The terminal output is as shown in the attached image.
Default synthesis process，the third line of output is:
[image]
Modified Synthesis， the third line of output is:
[
[yosys_1023.zip](https://github.com/YosysHQ/yosys/files/13070037/yosys_1023.zip)
]([image])
Please find attached the code of my program.
Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.

### Expected Behavior

Consistent simulation results

### Actual Behavior

Inconsistent simulation results

jix, 2023-10-23:
Given that rtl.v is either generated or obfuscated and not exactly small, I think there is little chance that someone will actually be able to debug this. I can verify that the simulation output differs, but given the unreadable source, I can't even tell whether the design is doing something where getting different synthesis results would be expected behavior or where there is some known synth vs simulation mismatch (either on the input or output side).

oomimo18, 2023-10-23:
> Given that rtl.v is either generated or obfuscated and not exactly small, I think there is little chance that someone will actually be able to debug this. I can verify that the simulation output [...]

Hello, jix. thank you for your response. After I submitted the bug report, I also analyzed this issue. I finally determined that there is a problem with the 'opt_expr -fine' pass because when I removed the 'opt_expr -fine' at the very end, the inconsistency in simulation results disappeared. It seems that the 'opt_expr -fine' pass is causing the inconsistency in simulation results. I hope you can verify this on your end.
Looking forward to receiving your response again.

whitequark, 2023-10-23:
You may be able to use the `bugpoint` pass to automatically minimize the netlist, if you can produce a way to compare synthesis/simulation results automatically as well.

georgerennie, 2025-06-06:
After fixing a bug that appeared in Yosys in the meantime (#5164), I was no longer able to reproduce this issue on current main - when I try simulating I get the same results for both output verilog files. Thus I assume this issue shadows one of the other fuzzer found issues that have been fixed in the last year and a half so I will close this.

closed 2025-06-06 by georgerennie