YosysHQ/yosys #4016: Multiple edge sensitive events found for this signal

abdulhameed-rs, 2023-11-02:
### Version

Yosys 0.18+10

### On which OS did this happen?

Linux

### Reproduction Steps

I tried running a design with this piece of code in it as given below:

always @(posedge clk or posedge rst or posedge start)
begin
	if(rst | start)
	begin
		writeOutputLine <= 0;
		writeNextValidLine <= 0;
		writeNextPlusOne <= 1;
		getNextPlusOne <= 1;
	end
	else
	begin
		if(writeRowCount >= writeNextValidLine) //When writeRowCount becomes higher than the next valid line to read out, comptue the next valid line.
		begin
			if(getNextPlusOne)			//Keep writeNextPlusOne
			begin
				writeNextPlusOne <= writeNextValidLine + 1;
			end
			getNextPlusOne <= 0;
			writeOutputLine <= writeOutputLine + 1;
			writeNextValidLine <= ((writeOutputLine*yScale + {{(OUTPUT_Y_RES_WIDTH + SCALE_INT_BITS){1'b0}}, topFracOffset}) >> SCALE_FRAC_BITS) & {{SCALE_BITS{1'b0}}, {OUTPUT_Y_RES_WIDTH{1'b1}}};
		end
		else
		begin
			getNextPlusOne <= 1;
		end
	end
end

It gives an error with multiple edge sensitive events found. Same design synthesizes perfectly on **Vivado**.
This piece of code works in yosys by creating intermidiate signal for "**rst | start**" and providing that signal to always block.
Is there anyway yosys can synthesize this code without creating intermediate signal?

### Expected Behavior

I expect it to be synthesized by logic or gates being used.

### Actual Behavior

3.8.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\test.\out' using process `\test.$proc$/Verilog_Cordic_Core/project_1/project_1.srcs/sources_1/test.v:8$2'.
ERROR: Multiple edge sensitive events found for this signal!

alaindargelas, 2023-12-13:
This problem is fixed here, by the systemverilog frontend:
https://github.com/alaindargelas/synlig/tree/stable_rs
This issue can be closed.

povik, 2023-12-14:
@synlig While it's good to point out the synlig frontend can handle this, it does not mean there's not an issue with the internal Yosys frontend. So I think the issue should stay, though maybe it's duplicate.

povik, 2023-12-14:
As far as I can tell this is a feature request for expanding of what Yosys considers synthesizable Verilog, classifying the issue as such.