YosysHQ/yosys #4034: Formal cells shouldn't be marked evaluable in `celltypes.h`

povik, 2023-11-16:
### Version

Yosys 0.34+23 (git sha1 59fbee400, clang++ 11.1.0 -fPIC -Os)

### On which OS did this happen?

Linux, macOS, Windows, BSD

### Reproduction Steps

Look into `kernel/celltypes.h` where formal cell types are registered, with the last argument to `setup_type` being whether the cell is evaluable (`CellTypes::cell_evaluable` then returns true for the cell).

https://github.com/YosysHQ/yosys/blob/27ac91270991ad7c2304e5f90a5c1001c2f08ec1/kernel/celltypes.h#L90-L99

### Expected Behavior

Formal cells like `$assert` are registered as non-evaluable since the cell can have a side effect. This depends on the interpretation of 'evaluable' of course, but I suggest we adopt the definition of "the cell is purely combinational and its output only depends on current inputs", which I think is the most useful definition for writing optimization passes.

### Actual Behavior

Formal cells are registered evaluable. Even if we go with a definition of 'evaluable' for which this is right, I suggest we switch `$print` to evaluable as well (currently it's non-evaluable), since it's very much like `$assert` so this would be the consistent thing to do. The more consistent `CellTypes::cell_evaluable` is, the more useful it is.

nakengelhardt, 2023-11-20:
While "combinatorial" is the practical definition of evaluable, isn't it technically "`ConstEval` / the `eval` pass knows what to do with this"?

povik, 2023-11-20:
> isn't it technically "`ConstEval` / the `eval` pass knows what to do with this"?

That's at least how the `sim` code is written (related #3913).

nakengelhardt, 2023-11-20:
I'm in favor of separating out the two use cases with a "is combinatorial" flag, but I'd like it to be located elsewhere (cells?) so that it's possible to set for blackboxes (would be very useful for show or scc).

povik, 2023-11-20:
> but I'd like it to be located elsewhere (cells?)

We can still put it in `CellTypes` and look for attributes on module definitions when `setup_design` is called. `CellTypes` already inspects the defined modules to detect inputs/outputs, this can be viewed an extension of that.

nakengelhardt, 2023-11-20:
I forgot `CellTypes` can do that! That's ideal then.

povik, 2023-11-20:
Ideally we would detect a module composed of combinational cells and carry over the combinational flag to the module cell type, but I don't see how to do that in both robust and performant way (this sounds like something easier to do in NGY).

povik, 2023-11-20:
(The struggle here is to maintain that information automatically, if it's a work of an explicit pass which revisits the modules and sets/clears the combinational attribute, that's doable.)

jix, 2023-11-20:
As it came up during discussing this issue, the definition I have been using for a combinational cell: A cell where the outputs are given by a deterministic function of the inputs, which in particular means the outputs do not depend on any state, time or solver choices.

nakengelhardt, 2023-11-20:
Action Items:
	1. add tests that check that all things marked evaluable do not crash when passed to eval
	2. adjust the evaluable flags so that the new test passes
	3. consider other refactoring with more flags/filtering by different flags etc.