YosysHQ/yosys #4039: [abc] Assertion src/base/abc/abcAig.c:1134: void abc::Abc_AigUpdateLevelR_int(abc::Abc_Aig_t *): Assertion `Abc_ObjIsNode(pNode)' failed.

dlmiles, 2023-11-19:
### Version

Yosys 0.33+21 (git sha1 9e004426e, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230913

### On which OS did this happen?

Linux

### Reproduction Steps

Test case ZIP attached.

FAIL: Yosys 0.33+21 (git sha1 9e004426e, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230913   FIRST FAILING VERSION

yosys-abc: src/base/abc/abcAig.c:1134: void abc::Abc_AigUpdateLevelR_int(abc::Abc_Aig_t *): Assertion `Abc_ObjIsNode(pNode)' failed.

----

It looks like ABC went:

https://github.com/YosysHQ/abc
bb64142..9537f39

In that time this assertion was triggered.

----

OpenLane flow emits:

```
executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: ../work/runs/wokwi/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:

67. Executing ABC pass (technology mapping using ABC).

67.1. Extracting gate netlist of module `\tt_um_dlmiles_tt05_i2c_bert' to `/tmp/yosys-abc-FpVpDJ/input.blif'..
Extracted 1431 gates and 1589 wires to a netlist network with 156 inputs and 123 outputs.

67.1.1. Executing ABC.
sh: line 1:   265 Aborted                 (core dumped) "/build/bin/yosys-abc" -s -f /tmp/yosys-abc-FpVpDJ/abc.script 2>&1
ERROR: ABC: execution of command ""/build/bin/yosys-abc" -s -f /tmp/yosys-abc-FpVpDJ/abc.script 2>&1" failed: return code 134.
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

### Expected Behavior

Yosys:
PASS: Yosys 0.28+6 (git sha1 cee3cb31b, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230504
PASS: Yosys 0.31+16 (git sha1 b04d0e09e, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230725
PASS: Yosys 0.32+79 (git sha1 b75959f1f, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230905
PASS: Yosys 0.33+3 (git sha1 83b1a57ee, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230906
PASS: Yosys 0.33+6 (git sha1 41b34a193, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230911
PASS: Yosys 0.33+6 (git sha1 41b34a193, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230912

yosys-abc: src/base/abc/abcAig.c:1134: void abc::Abc_AigUpdateLevelR_int(abc::Abc_Aig_t *): Assertion `Abc_ObjIsNode(pNode)' failed.

FAIL: Yosys 0.33+21 (git sha1 9e004426e, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230913   FIRST FAILING VERSION
FAIL: Yosys 0.33+53 (git sha1 e2b613355, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20230919
FAIL: Yosys 0.33+112 (git sha1 3e02b63ee, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20231005
FAIL: Yosys 0.34+7 (git sha1 fc815fdb4, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20231006
FAIL: Yosys 0.34+60 (git sha1 93a426cbb, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20231107
FAIL: Yosys 0.35+7 (git sha1 5691cd095, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20231108
FAIL: Yosys 0.35+29 (git sha1 ab6c1d368, clang 10.0.0-4ubuntu1 -fPIC -Os)  oss-cad-suite-20231118

### Actual Behavior

```
⦗OSS CAD Suite⦘ $ projects/yosys-abc-FpVpDJ/run.sh
ABC command line: "source abc.script".

+ read_blif /tmp/yosys-abc-FpVpDJ/input.blif 
+ read_lib -w /tmp/yosys-abc-FpVpDJ/work/runs/wokwi/tmp/synthesis/trimmed.lib 
Parsing finished successfully.  Parsing time =     0.04 sec
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfbbn_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfrbp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfrtp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfrtp_4".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfsbp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfstp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfstp_4".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfxbp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfxtp_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dfxtp_4".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dlxtn_1".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dlxtn_2".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dlxtn_4".
Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hd__dlxtp_1".
Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hd__ebufn_2".
Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hd__ebufn_4".
Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hd__ebufn_8".
Library "sky130A_merged" from "/tmp/yosys-abc-FpVpDJ/work/runs/wokwi/tmp/synthesis/trimmed.lib" has 175 cells (17 skipped: 14 seq; 3 tri-state; 0 no func; 0 dont_use).  Time =     0.07 sec
Memory =    9.54 MB. Time =     0.07 sec
Warning: Detected 2 multi-output gates (for example, "sky130_fd_sc_hd__fa_1").
+ read_constr -v /tmp/yosys-abc-FpVpDJ/work/runs/wokwi/tmp/synthesis/synthesis.sdc 
Setting driving cell to be "sky130_fd_sc_hd__inv_2".
Setting output load to be 33.442001.
+ read_constr /tmp/yosys-abc-FpVpDJ/work/runs/wokwi/tmp/synthesis/synthesis.sdc 
+ fx 
+ mfs 
+ strash 
+ refactor 
yosys-abc: src/base/abc/abcAig.c:1134: void abc::Abc_AigUpdateLevelR_int(abc::Abc_Aig_t *): Assertion `Abc_ObjIsNode(pNode)' failed.
Aborted (core dumped)
```

dlmiles, 2023-11-19:
[yosys-abc-FpVpDJ-issue4039.tar.bz2.gz](https://github.com/YosysHQ/yosys/files/13403813/yosys-abc-FpVpDJ-issue4039.tar.bz2.gz)

Only BZIP2 can get it below the github 25MB limit.

github does not accept tar.bz2 attachments, so it is wrapped in gzip.

```$ gunzip -c yosys-abc-FpVpDJ-issue4039.tar.bz2.gz | tar -xtvf - -C /tmp && cd /tmp/yosys-abc-FpVpDJ && ls -l
$ gunzip -c yosys-abc-FpVpDJ-issue4039.tar.bz2.gz | tar -jtvf -
drwx------ dlm/dlm           0 2023-11-19 16:15 yosys-abc-FpVpDJ/
-rw-r--r-- dlm/dlm         754 2023-11-19 14:12 yosys-abc-FpVpDJ/stdcells.genlib
-rw-r--r-- dlm/dlm        2485 2023-11-19 14:17 yosys-abc-FpVpDJ/abc.script.orig
-rwxr-xr-x dlm/dlm         858 2023-11-19 16:07 yosys-abc-FpVpDJ/run.sh
-rw-r--r-- dlm/dlm      124067 2023-11-19 14:12 yosys-abc-FpVpDJ/input.blif
-rw-r--r-- dlm/dlm        2611 2023-11-19 14:18 yosys-abc-FpVpDJ/abc.script
drwxrwxr-x dlm/dlm           0 2023-11-19 14:19 yosys-abc-FpVpDJ/work/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/openlane/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/openlane/scripts/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/openlane/scripts/yosys/
-rw-r--r-- dlm/dlm       15892 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/openlane/scripts/yosys/synth.tcl
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/
-rw-r--r-- dlm/dlm     6316524 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/1-trimmed.no_pg.lib
-rw-r--r-- dlm/dlm     6293484 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/trimmed.lib
-rw-r--r-- dlm/dlm          56 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/synthesis.sdc
-rw-r--r-- dlm/dlm        6977 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/trimmed.lib.exclude.list
-rw-r--r-- dlm/dlm    12893729 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/tmp/synthesis/1-sky130_fd_sc_hd__tt_025C_1v80.no_pg.lib
-rwxr-xr-x dlm/dlm        2380 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/run.sh
-rw-r--r-- dlm/dlm        2049 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/env.lldb
-rw-r--r-- dlm/dlm        2185 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/env.gdb
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/
-rw-r--r-- dlm/dlm        8064 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/csa_map.v
-rw-r--r-- dlm/dlm         390 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v
-rw-r--r-- dlm/dlm         415 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/latch_map.v
-rw-r--r-- dlm/dlm         139 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/tribuff_map.v
-rw-r--r-- dlm/dlm        2060 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/pdk/sky130A/libs.tech/openlane/sky130_fd_sc_hd/rca_map.v
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/
-rw-r--r-- dlm/dlm        4472 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/top_tt_um_dlmiles_tt05_i2c_bert.v
-rw-r--r-- dlm/dlm         110 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/config.vh
-rw-r--r-- dlm/dlm         156 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/tt_um.vh
drwxr-xr-x dlm/dlm           0 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/pdk_sky130/
-rw-r--r-- dlm/dlm        1165 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/pdk_sky130/sky130_fd_sc_hd__dlrtp.v
-rw-r--r-- dlm/dlm        1102 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/pdk_sky130/sky130_fd_sc_hd__maj3.v
-rw-r--r-- dlm/dlm       67200 2023-11-19 14:12 yosys-abc-FpVpDJ/work/runs/wokwi/issue_reproducible/work/src/TT05I2CBertTop.v
[... cut ...]
-rw-rw-r-- dlm/dlm          51 2023-11-18 23:35 yosys-abc-FpVpDJ/work/runs/wokwi/PDK_SOURCES
-rw-rw-r-- dlm/dlm       64551 2023-11-19 15:37 yosys-abc-FpVpDJ/output.blif
total 208
-rw-r--r--. 1 dlm dlm   2611 Nov 19 14:18 abc.script
-rw-r--r--. 1 dlm dlm   2485 Nov 19 14:17 abc.script.orig
-rw-r--r--. 1 dlm dlm 124067 Nov 19 14:12 input.blif
-rw-rw-r--. 1 dlm dlm  64551 Nov 19 15:37 output.blif
-rwxr-xr-x. 1 dlm dlm    858 Nov 19 16:07 run.sh
-rw-r--r--. 1 dlm dlm    754 Nov 19 14:12 stdcells.genlib
drwxrwxr-x. 3 dlm dlm   4096 Nov 19 14:19 work
```

See `run.sh`

jix, 2023-11-19:
Note that Yosys itself doesn't use the crashing `refactor` command. Given that the script uses the obsolete known-broken `rewrite` command which got replaced by `drw` as of 2016 (see https://github.com/YosysHQ/yosys/issues/174#issuecomment-222353448 and https://github.com/The-OpenROAD-Project/OpenLane/issues/1523) and given that there is a `drf` command that also performs refactoring and follows the same naming scheme as `drw`, my guess would be that `refactor` is also considered obsolete by upstream abc and has bit-rotted.

For this particular example, replacing all uses of `refactor` with `drf` and `rewrite` with `drw` seems to work, although I cannot tell whether they are adequate replacements for OpenLane's use case in general.

If this issue indeed only affects obsolete commands with working replacements, my suggestion would be that we remove such commands from yosys-abc, replacing them with an error message pointing at the replacement command to make sure that third party flows do not inadvertently keep using broken abc commands.

The alternative would be to get this fixed in upstream abc, but given the amount of effort it takes to debug issues in abc, the fact that there seems to be a working drop-in replacement with `drf` and the fact that Yosys itself does not use the `refactor` abc command, this might not happen anytime soon.