YosysHQ/yosys #4056: Synthesis Premature Termination and Synthesis Failure

DeLoSoCode, 2023-12-06:
### Version

 0.35+56

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,
I am using Yosys to read RTL code files and perform synthesis. However, during the synthesis process, Yosys encounters a crash issue, preventing the completion of synthesis (as shown in the screenshot below).
[image]
Synthesis processes as follows:
read_verilog rtl.v         
synth
write_verilog syn_yosys.v

Please find attached the code of RTL and and the synthesis log.
Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.
[yosys_project.zip](https://github.com/YosysHQ/yosys/files/13583429/yosys_project.zip)

### Expected Behavior

synthesis  success

### Actual Behavior

synthesis  fail

jix, 2023-12-06:
Note that this doesn't crash, but rather hangs without producing output (the screenshot also doesn't show a crash). When I enable debug output using `-g` I can see that it still makes progress, but every sub-pass takes very long to complete and the memory usage is close to 60GB. I can also see from the debug output that there are signals that are over 200000 bit wide, but it's hard to tell whether overall this is expected performance or not since rtl.v is fuzzer generated code that does not seem to be minimized and that I find quite unreadable.

jix, 2023-12-06:
I let it running and it did finish successfully after 37 minutes while staying below 60GB:

```
=== top ===

   Number of wires:               1586
   Number of wire bits:           5364
   Number of public wires:         155
   Number of public wire bits:    3589
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               1974
     $_ANDNOT_                     369
     $_AND_                         24
     $_DFFE_PN_                     13
     $_DFFE_PP_                     35
     $_DFF_P_                      237
     $_MUX_                        197
     $_NAND_                        55
     $_NOR_                         44
     $_NOT_                         95
     $_ORNOT_                       65
     $_OR_                         479
     $_SDFFCE_PN0N_                 16
     $_SDFFCE_PP1P_                  1
     $_SDFFE_PP0P_                   4
     $_SDFF_PN0_                    45
     $_SDFF_PN1_                     3
     $_SDFF_PP0_                    28
     $_XNOR_                        61
     $_XOR_                        203
```

The output certainly isn't large enough to justify that runtime, but the rtl still might generate logic that is large enough to justify this, just that it is eventually all optimized away, so I still can't say whether this should be a considered an actual performance bug.

DeLoSoCode, 2023-12-06:
> I left it running and it did finish successfully after 37 minutes while staying below 60GB:
> 
> ```
> === top ===
> 
>    Number of wires:               1586
>    Number of wire bits:           5364
>    Number of public wires:         155
>    Number of public wire bits:    3589
>    Number of memories:               0
>    Number of memory bits:            0
>    Number of processes:              0
>    Number of cells:               1974
>      $_ANDNOT_                     369
>      $_AND_                         24
>      $_DFFE_PN_                     13
>      $_DFFE_PP_                     35
>      $_DFF_P_                      237
>      $_MUX_                        197
>      $_NAND_                        55
>      $_NOR_                         44
>      $_NOT_                         95
>      $_ORNOT_                       65
>      $_OR_                         479
>      $_SDFFCE_PN0N_                 16
>      $_SDFFCE_PP1P_                  1
>      $_SDFFE_PP0P_                   4
>      $_SDFF_PN0_                    45
>      $_SDFF_PN1_                     3
>      $_SDFF_PP0_                    28
>      $_XNOR_                        61
>      $_XOR_                        203
> ```
> 
> The output certainly isn't large enough to justify that runtime, but the rtl still might generate logic that is large enough to justify this, just that it is eventually all optimized away, so I still can't say whether this should be a considered an actual performance bug.

Thank you very much for your response. I synthesized this RTL design file using the Vivado synthesis tool, and it quickly generated the corresponding synthesis files. However, as you mentioned, obtaining synthesis results quickly with the Yosys tool is challenging. Therefore, I strongly agree with your statement, this is a performance bug. The attached files below include the synthesis files and synthesis log generated using the Vivado tool. I look forward to your response. Thank you.
[vivado_test.zip](https://github.com/YosysHQ/yosys/files/13587702/vivado_test.zip)

jix, 2023-12-06:
> However, as you mentioned, obtaining synthesis results quickly with the Yosys tool is challenging. Therefore, I strongly agree with your statement, this is a performance bug.

That is not at all what I said, I said that I could not tell whether it is a performance bug since I found rtl.v unreadable and hence considered it a possibility that rtl.v describes large amounts of redundant logic in which case bad performance would have been expected.

I did some more debugging, though, and it did turn out to be caused by a bug in `peepopt`.

commit 7b74caa5 (2023-12-07): peepopt: Fix padding for the peepopt_shiftmul_right pattern

The previous version could easily generate a large amount of padding
when the constant factor was significantly larger than the width of the
shift data input. This could lead to huge amounts of logic being
generated before then being optimized away at a huge performance and
memory cost.

Additionally and more critically, when the input width was not a
multiple of the constant factor, the input data was padded with 'x bits
to such a multiple before interspersing the 'x padding needed to align
the selectable windows to power-of-two off [... cut]
closed 2023-12-07 by povik