YosysHQ/yosys #4077: Yosys Verilog Parsing Error: Issue in AST Generation

DeLoSoCode, 2023-12-14:
### Version

yosys 0.35+56

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,
While attempting synthesis using Yosys, I encountered an issue. The command read_verilog rtl.v resulted in the error message shown in the attached screenshot.
[image]
The design file (rtl.v) successfully undergoes synthesis using Vivado synthesis tools, indicating that the problem seems to be unique to Yosys. 

Please find attached the code of RTL .
Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.
[project_one.zip](https://github.com/YosysHQ/yosys/files/13673108/project_one.zip)

### Expected Behavior

synthesis success

### Actual Behavior

synthesis fails

jix, 2023-12-14:
This is a minimized reproducer for that assertion, changing the size of `a` to a non-multiple of 8 doesn't trigger the assertion:

```
module top(input clk);
    reg [7:0] a = '0;
    always @(posedge clk)
        a <= {"X", 1 ? "" : a};
endmodule
```

DeLoSoCode, 2023-12-17:
> This is a minimized reproducer for that assertion, changing the size of `a` to a non-multiple of 8 doesn't trigger the assertion:
> 
> ```
> module top(input clk);
>     reg [7:0] a = '0;
>     always @(posedge clk)
>         a <= {"X", 1 ? "" : a};
> endmodule
> ```

Thank you very much for your help