YosysHQ/yosys #4079: Error Arises During Synthesis due to Verilog Code Structure

DeLoSoCode, 2023-12-15:
### Version

yosys 0.35+56

### On which OS did this happen?

Linux

### Reproduction Steps

Hello,
While engaging in the synthesis process with Yosys, I faced a challenge when read the design file (rtl.v). Executing the read_verilog rtl.v command led to the error message captured in the attached screenshot. This occurred during attempts to optimize and synthesize the design using Yosys tools.
[image]
Synthesis processes as follows:
read_verilog rtl.v         
synth
write_verilog syn_yosys.v

The synthesis of the design file (rtl.v) is successful when employing Vivado synthesis tools. This success suggests that the issue is specific to Yosys. 

Please find attached the code of RTL .
Thank you in advance for your attention to this matter.
I look forward to hearing from you regarding this issue.
[yosys_project_1215.zip](https://github.com/YosysHQ/yosys/files/13680815/yosys_project_1215.zip)

### Expected Behavior

synthesis success

### Actual Behavior

synthesis fail

povik, 2024-01-08:
Minimal reproducer:
```
module top;
	wire [7:0] w = {"a", (1 ? "b" : "cc")};
endmodule
```

povik, 2024-01-08:
Possibly relevant: https://github.com/YosysHQ/yosys/commit/891e4b5b0d16a537e6d1ca0ecbc1ac3bafccecef

DeLoSoCode, 2024-01-12:
> Minimal reproducer:
> 
> ```
> module top;
> 	wire [7:0] w = {"a", (1 ? "b" : "cc")};
> endmodule
> ```

Thank you very much for your help