YosysHQ/yosys #4096: Can't call DPI function with no return (void return type)

punzik, 2023-12-29:
### Version

Yosys 0.36 (git sha1 8f07a0d8404, gcc 12.3.0 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

test.sv
```verilog
module test;
    import "DPI-C" function void test (integer);

    initial begin
        test(123);
    end
endmodule
```

test.c
```c
void test(int x)
{
    printf("%i\n", x);
}
```

test.ys
```
plugin -i test.so
read -sv2012 test.sv
hierarchy -check -top test
```

Makefile
```
all:
	gcc -shared -o test.so test.c
	yosys test.ys
```

### Expected Behavior

Calling the `test()` function during elaboration.

### Actual Behavior

When trying to call the C function `void test(int)`, the error `ERROR: Can't resolve task name \test'` occurs. 
If the function `test` returns a value, yosys runs without error.

povik, 2024-01-08:
Could you please post reproduction steps for the working case? I get ```test.sv:5: ERROR: Can't resolve task name `\test'.``` either way.

punzik, 2024-01-13:
To get a working case, you need to fix the test.sv file:

```verilog
module test;
    import "DPI-C" function integer test (integer);

    initial begin
        integer unused;
        unused = test(123);
    end
endmodule
```

The C code can be left as it is (but add `#include <stdio.h>`).