YosysHQ/yosys #4107: ERROR: Unsupported internal cell `$pow'

flaviens, 2024-01-05:
### Version

Yosys 0.29+11 (git sha1 0b05a6425, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I've noticed an issue with the $pow cell for CXXRTL:

`top.v`

```
 module top(
    input  a,
    input  b,
    output y
);

    assign y = a ** b;
endmodule
```

Then execute:

```
yosys -p "read_verilog -sv top.v; write_cxxrtl top.cpp"
```

Thanks!

### Expected Behavior

Successful Yosys execution.

### Actual Behavior

Yosys says `ERROR: Unsupported internal cell `$pow'.`

whitequark, 2024-01-05:
What the error message means is that the `**` operation is not implemented in CXXRTL. At the moment I do not plan to implement it but I will accept a PR implementing it.

flaviens, 2024-01-05:
Hi Whitequark, thank you for your quick reply! It's not the most common cell indeed :) I let you close the issue if you don't want to keep it open.
Thanks!

closed 2024-01-05 by whitequark