YosysHQ/yosys #4112: Bug with modulo in CXXRTL when operands intersect

flaviens, 2024-01-06:
### Version

Yosys 0.29+11 (git sha1 0b05a6425, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed that the modulo operation sometimes provides wrong results in CXXRTL when there is an intersection between the operands.

For example, this design, supplied with `in_data=0x216231b1f16e9e8`:
```
module top(
    input logic [63:0] in_data,
    output logic [63:0] out_data
);
    assign out_data = in_data[41:6] % in_data[63:28];
endmodule
```

Some valuations of `in_data` yield a correct result and seem to compute faster than faulty valuations.

When there is no intersection between the operands, it seems that the calculation is correct, for example:
```
module top(
    input logic [63:0] in_data,
    output logic [63:0] out_data
);
    assign out_data = in_data[63:32] % in_data[31:0];
endmodule
```

Thanks!
Flavien

### Expected Behavior

Output value: `0x35522118`

### Actual Behavior

Output value: `0x90beaf8`

whitequark, 2024-01-06:
You must post the complete reproduction steps, not just your Verilog code.

flaviens, 2024-01-06:
Hi @whitequark, thank you for your quick response! I adapted the post according to your feedback. Please let me know if you would like more information!

povik, 2024-01-08:
I think `0x90beaf8` is expected, at least that's what I get with
```
yosys -p "read_verilog -sv top.sv; eval -set in_data 64'h216231b1f16e9e8 w:out_data"
```
Though there's still a discrepancy, since CXXRTL gives me `0x435522118`.

whitequark, 2024-01-19:
@flaviens I cannot reproduce this, and the condition ("intersection between operands") doesn't even make sense. `0x90beaf8` is the correct result.

```
>>> hex((0x216231b1f16e9e8>>6) & ((1 << (41-6+1))-1))
'0xc6c7c5ba7'
>>> hex((0x216231b1f16e9e8>>28) & ((1 << (63-28+1))-1))
'0x216231b1'
>>> hex(0xc6c7c5ba7 % 0x216231b1)
'0x90beaf8'
```

whitequark, 2024-01-20:
This was probably fixed by commit https://github.com/YosysHQ/yosys/commit/ded63bedd589255e6e06747edcdbf262444ac379 and/or https://github.com/YosysHQ/yosys/commit/ff53f3d2b6e267358cf4c3d08574416c79441ef9 and/or https://github.com/YosysHQ/yosys/commit/d7cb6981b55a15eee3d0d9be84446826a65c0403.

closed 2024-01-20 by whitequark