YosysHQ/yosys #4113: Bug: Logic and arithmetic left shifts sometimes overflow the output signal

flaviens, 2024-01-06:
### Version

Yosys 0.29+11 (git sha1 0b05a6425, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I encountered an issue for both logic and arithmetic left shifts. In some cases, it looks that the output may overflow.
For example, you can take this design:
```
module top(input logic [31:0] a, input logic [31:0] b, output logic [4:0] out_data);
  assign out_data = b[5:1] << a[31:27];
endmodule
```

And a testbench like:
```
int main() {
  uint32_t a = 0x4e57633b;
  uint32_t b = 0x468e7520;
  top.p_a.set<uint32_t>(a);
  top.p_b.set<uint32_t>(b);
  top.step();
  uint32_t out_data = top.p_out__data.get<uint32_t>();
  std::cout << "Out data: " << std::hex << out_data << std::endl;
}
```

Thank you!
Flavien

### Expected Behavior

Out data: `0`

### Actual Behavior

Out data: `2000` (hex)

whitequark, 2024-01-06:
This behavior was intentional, but I agree that it is a bug.

whitequark, 2024-01-06:
> Yosys 0.29+11 

Do note that you are six Yosys releases behind.

povik, 2024-01-06:
> This behavior was intentional, but I agree that it is a bug.

Please elaborate. What aspect to the behavior was intentional?

flaviens, 2024-01-06:
> > Yosys 0.29+11
> 
> Do note that you are six Yosys releases behind.

Thanks for the tip! I re-tested and updated the post accordingly.

whitequark, 2024-01-06:
> What aspect to the behavior was intentional?

CXXRTL itself makes sure to mask the high bits of values at the point of use, rather than the point of computation, to avoid excessive and repeated masking. But `.get()` was never taught about that aspect. (This gets even worse when using the C API, so while `.get()` can and should be taught to clear the high bits, it may be a good idea to mask at the point of computation, so that consumers of the C API aren't burdened with this.)

povik, 2024-01-06:
> CXXRTL itself makes sure to mask the high bits of values at the point of use, rather than the point of computation, to avoid excessive and repeated masking.

Huh, I thought values are supposed to be masked at the point of computation already, with that in mind I wrote e.g. #4075. So we are in the middle of a transition?

whitequark, 2024-01-06:
> So we are in the middle of a transition?

Functionally yes. There is no harm in masking twice, so I've decided a while ago to blanket approve any changes that add masing at the point of computation. I didn't communicate it mainly due to the lack of time.

whitequark, 2024-01-06:
@povik Do you have any appetite for adding the masking to `.get` and `.set` at least?

whitequark, 2024-01-19:
@flaviens Not able to reproduce on commit fc5ff7a265293099e25812afbbbe8e687f0b62e8 on Linux x86_64 with clang++ or g++ `-O0` or `-O3`.

whitequark, 2024-01-20:
This was fixed in commit ff53f3d2b6e267358cf4c3d08574416c79441ef9, I think.

closed 2024-01-20 by whitequark