YosysHQ/yosys #4114: Bug: incorrect division

flaviens, 2024-01-06:
### Version

Yosys 0.29+11 (git sha1 0b05a6425, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed that the division operation sometimes provides wrong results in CXXRTL. It seems to require an intersection between the operands, like #4112 .

For example, this design, supplied with `in_data=0x32fff902684a481a`:
```
module top(
    input logic [63:0] in_data,
    output logic [63:0] out_data
);
    assign out_data = in_data[41:6] / in_data[63:28];
endmodule
```

Some valuations of `in_data` yield a correct result and seem to compute faster than faulty valuations.

When there is no intersection between the operands, it seems that the calculation is correct, for example:
So there's strong indications that the root cause is similar to #4112 .

Thanks!
Flavien

### Expected Behavior

Out data: `1`

### Actual Behavior

Out data: `ffffffff`

povik, 2024-01-08:
Wow, this is slow indeed.

whitequark, 2024-01-08:
> It seems to require an intersection between the operands, like #4112 .

I think this happens because there is an `opt_expr` pass at the end of `proc`. If you call `-noproc` it should not require any such intersection.

whitequark, 2024-01-19:
@flaviens I cannot reproduce this on commit fc5ff7a265293099e25812afbbbe8e687f0b62e8.

whitequark, 2024-01-20:
This was probably fixed by commit ded63bedd589255e6e06747edcdbf262444ac379 and/or ff53f3d2b6e267358cf4c3d08574416c79441ef9 and/or d7cb6981b55a15eee3d0d9be84446826a65c0403.

closed 2024-01-20 by whitequark