YosysHQ/yosys #4115: Bug: Shifts (in both directions) consider the operand signed under some conditions in CXXRTL

flaviens, 2024-01-07:
### Version

Yosys 0.29+11 (git sha1 f7ffd12bf, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I noticed a bug in CXXRTL, that could be reduced to the occurrence below.

#### Design

For example, consider this design, supplied with `c=0x40e831aa`:
```
module top(
input logic [31:0] c,

output logic [31:0] out,
);
  // assign out = c >> 28'hfffffff8;
  assign out = c >> 32'hfffffff8;
endmodule
```

The commented and the uncommented line should produce the same result: pushing c super far to the right direction, effectively producing only zeroes.
However, I noticed that when the offset is encoded on at least 32 bits (there might be more conditions that I did not yet notice), the offset is considered signed.

#### Testbench

The testbench can look like this: `tb_cxxrtl.cc`:

```
#include "top.cpp"
#include <iostream>
int main(int argc, char **argv, char **env) {
  cxxrtl_design::p_top top;
  uint32_t c = 0x40e831aa;
  top.p_c.set<uint32_t>(c);
  top.step();
  uint64_t out = top.p_out.get<uint64_t>();
  std::cout << "Out data: " << std::hex << out << std::endl;
  exit(0);
}

```

#### Commands

```
yosys -p "read_verilog -sv top.sv; write_cxxrtl top.cpp"
g++ -g -O3 -std=c++14 -I `yosys-config --datdir`/include tb_cxxrtl.cc -o tb
./tb
```

Thanks!
Flavien

### Expected Behavior

Out data: `0`

### Actual Behavior

Out data: `e831aa00` (hex)

whitequark, 2024-01-19:
Cannot reproduce on commit fc5ff7a265293099e25812afbbbe8e687f0b62e8.

whitequark, 2024-01-20:
This was fixed in commits ded63bedd589255e6e06747edcdbf262444ac379 and ff53f3d2b6e267358cf4c3d08574416c79441ef9, I think.

closed 2024-01-20 by whitequark