YosysHQ/yosys #4116: Bug: Segmentation fault during step for CXXRTL

flaviens, 2024-01-07:
### Version

Yosys 0.29+11 (git sha1 f7ffd12bf, gcc 7.5.0-6ubuntu2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi there!

I encountered a segmentation fault with cxxrtl during the evaluation step.

The gdb backtrace is
```
Program received signal SIGSEGV, Segmentation fault.
0x000055555556d132 in cxxrtl::value<273ul>::blit<0ul, 0ul> (source=..., this=0x0) at /path/to/cxxrtl.h:254
254             value<Bits> blit(const value<Stop - Start + 1> &source) const {
(gdb) bt
#0  0x000055555556d132 in cxxrtl::value<273ul>::blit<0ul, 0ul> (source=..., this=0x0) at /path/to/cxxrtl.h:254
#1  cxxrtl::slice_expr<cxxrtl::value<273ul>, 0ul, 0ul>::operator= (rhs=..., this=0x7fffffffb838) at /path/to/cxxrtl.h:603
#2  _ZN6cxxrtl11concat_exprINS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS_10slice_exprINS_5valueILm273EEELm272ELm253EEENS1_INS2_ILm96EEELm63ELm37EEEEENS1_IS3_Lm98ELm98EEEEENS1_IS3_Lm48ELm48EEEEENS1_IS3_Lm97ELm97EEEEENS1_IS3_Lm47ELm47EEEEENS1_IS3_Lm96ELm96EEEEENS1_IS3_Lm46ELm46EEEEENS1_IS3_Lm95ELm95EEEEENS1_IS3_Lm45ELm45EEEEENS1_IS3_Lm94ELm94EEEEENS1_IS3_Lm44ELm44EEEEENS1_IS3_Lm93ELm93EEEEENS1_IS3_Lm43ELm43EEEEENS1_IS3_Lm92ELm92EEEEENS1_IS3_Lm42ELm42EEEEENS1_IS3_Lm91ELm91EEEEENS1_IS3_Lm41ELm41EEEEENS1_IS3_Lm90ELm90EEEEENS1_IS3_Lm40ELm40EEEEENS1_IS3_Lm89ELm89EEEEENS1_IS3_Lm39ELm39EEEEENS1_IS3_Lm88ELm88EEEEENS1_IS3_Lm38ELm38EEEEENS1_IS3_Lm87ELm87EEEEENS1_IS3_Lm37ELm37EEEEENS1_IS3_Lm86ELm86EEEEENS1_IS3_Lm36ELm36EEEEENS1_IS3_Lm85ELm85EEEEENS1_IS3_Lm35ELm35EEEEENS1_IS3_Lm84ELm84EEEEENS1_IS3_Lm34ELm34EEEEENS1_IS3_Lm83ELm83EEEEENS1_IS3_Lm33ELm33EEEEENS1_IS3_Lm82ELm82EEEEENS1_IS3_Lm32ELm32EEEEENS1_IS3_Lm81ELm81EEEEENS1_IS3_Lm31ELm31EEEEENS1_IS3_Lm80ELm80EEEEENS1_IS3_Lm30ELm30EEEEENS1_IS3_Lm79ELm79EEEEENS1_IS3_Lm29ELm29EEEEENS1_IS3_Lm78ELm78EEEEENS1_IS3_Lm28ELm28EEEEENS1_IS3_Lm77ELm77EEEEENS1_IS3_Lm27ELm27EEEEENS1_IS3_Lm76ELm76EEEEENS1_IS3_Lm26ELm26EEEEENS1_IS3_Lm75ELm75EEEEENS1_IS3_Lm25ELm25EEEEENS1_IS3_Lm74ELm74EEEEENS1_IS3_Lm24ELm24EEEEENS1_IS3_Lm73ELm73EEEEENS1_IS3_Lm23ELm23EEEEENS1_IS3_Lm72ELm72EEEEENS1_IS3_Lm22ELm22EEEEENS1_IS3_Lm71ELm71EEEEENS1_IS3_Lm21ELm21EEEEENS1_IS3_Lm70ELm70EEEEENS1_IS3_Lm20ELm20EEEEENS1_IS3_Lm69ELm69EEEEENS1_IS3_Lm19ELm19EEEEENS1_IS3_Lm68ELm68EEEEENS1_IS3_Lm18ELm18EEEEENS1_IS3_Lm67ELm67EEEEENS1_IS3_Lm17ELm17EEEEENS1_IS3_Lm66ELm66EEEEENS1_IS3_Lm16ELm16EEEEENS1_IS3_Lm65ELm65EEEEENS1_IS3_Lm15ELm15EEEEENS1_IS3_Lm64ELm64EEEEENS1_IS3_Lm14ELm14EEEEENS1_IS3_Lm63ELm63EEEEENS1_IS3_Lm13ELm13EEEEENS1_IS3_Lm62ELm62EEEEENS1_IS3_Lm12ELm12EEEEENS1_IS3_Lm61ELm61EEEEENS1_IS3_Lm11ELm11EEEEENS1_IS3_Lm60ELm60EEEEENS1_IS3_Lm10ELm10EEEEENS1_IS3_Lm59ELm59EEEEENS1_IS3_Lm9ELm9EEEEENS1_IS3_Lm58ELm58EEEEENS1_IS3_Lm8ELm8EEEEENS1_IS3_Lm57ELm57EEEEENS1_IS3_Lm7ELm7EEEEENS1_IS3_Lm56ELm56EEEEENS1_IS3_Lm6ELm6EEEEENS1_IS3_Lm55ELm55EEEEENS1_IS3_Lm5ELm5EEEEENS1_IS3_Lm54ELm54EEEEENS1_IS3_Lm4ELm4EEEEENS1_IS3_Lm53ELm53EEEEENS1_IS3_Lm3ELm3EEEEENS1_IS3_Lm52ELm52EEEEENS1_IS3_Lm2ELm2EEEEENS1_IS3_Lm51ELm51EEEEENS1_IS3_Lm1ELm1EEEEENS1_IS3_Lm50ELm50EEEEENS1_IS3_Lm0ELm0EEEEaSERKNS2_ILm145EEE (rhs=..., this=0x7fffffffbe80) at /path/to/cxxrtl.h:637
#3  _ZN6cxxrtl11concat_exprINS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS_10
slice_exprINS_5valueILm273EEELm272ELm253EEENS1_INS2_ILm96EEELm63ELm37EEEEENS1_IS3_Lm98ELm98EEEEENS1_IS3_Lm48ELm48EEEEENS1_IS3_Lm97ELm97EEEEENS1_IS3_Lm47ELm47EEEEENS1_IS3_Lm96ELm96EEEEENS1_IS3_Lm46ELm46EEEEENS1_IS3_Lm95ELm95EEEEENS1_IS3_Lm45ELm45EEEEENS1_IS3_Lm94ELm94EEEEENS1_IS3_Lm44ELm44EEEEENS1_IS3_Lm93ELm93EEEEENS1_IS3_Lm43ELm43EEEEENS1_IS3_Lm92ELm92EEEEENS1_IS3_Lm42ELm42EEEEENS1_IS3_Lm91ELm91EEEEENS1_IS3_Lm41ELm41EEEEENS1_IS3_Lm90ELm90EEEEENS1_IS3_Lm40ELm40EEEEENS1_IS3_Lm89ELm89EEEEENS1_IS3_Lm39ELm39EEEEENS1_IS3_Lm88ELm88EEEEENS1_IS3_Lm38ELm38EEEEENS1_IS3_Lm87ELm87EEEEENS1_IS3_Lm37ELm37EEEEENS1_IS3_Lm86ELm86EEEEENS1_IS3_Lm36ELm36EEEEENS1_IS3_Lm85ELm85EEEEENS1_IS3_Lm35ELm35EEEEENS1_IS3_Lm84ELm84EEEEENS1_IS3_Lm34ELm34EEEEENS1_IS3_Lm83ELm83EEEEENS1_IS3_Lm33ELm33EEEEENS1_IS3_Lm82ELm82EEEEENS1_IS3_Lm32ELm32EEEEENS1_IS3_Lm81ELm81EEEEENS1_IS3_Lm31ELm31EEEEENS1_IS3_Lm80ELm80EEEEENS1_IS3_Lm30ELm30EEEEENS1_IS3_Lm79ELm79EEEEENS1_IS3_Lm29ELm29EEEEENS1_IS3_Lm78ELm78EEEEENS1_IS3_Lm28ELm28EEEEENS1_IS3_Lm77ELm77EEEEENS1_IS3_Lm27ELm27EEEEENS1_IS3_Lm76ELm76EEEEENS1_IS3_Lm26ELm26EEEEENS1_IS3_Lm75ELm75EEEEENS1_IS3_Lm25ELm25EEEEENS1_IS3_Lm74ELm74EEEEENS1_IS3_Lm24ELm24EEEEENS1_IS3_Lm73ELm73EEEEENS1_IS3_Lm23ELm23EEEEENS1_IS3_Lm72ELm72EEEEENS1_IS3_Lm22ELm22EEEEENS1_IS3_Lm71ELm71EEEEENS1_IS3_Lm21ELm21EEEEENS1_IS3_Lm70ELm70EEEEENS1_IS3_Lm20ELm20EEEEENS1_IS3_Lm69ELm69EEEEENS1_IS3_Lm19ELm19EEEEENS1_IS3_Lm68ELm68EEEEENS1_IS3_Lm18ELm18EEEEENS1_IS3_Lm67ELm67EEEEENS1_IS3_Lm17ELm17EEEEENS1_IS3_Lm66ELm66EEEEENS1_IS3_Lm16ELm16EEEEENS1_IS3_Lm65ELm65EEEEENS1_IS3_Lm15ELm15EEEEENS1_IS3_Lm64ELm64EEEEENS1_IS3_Lm14ELm14EEEEENS1_IS3_Lm63ELm63EEEEENS1_IS3_Lm13ELm13EEEEENS1_IS3_Lm62ELm62EEEEENS1_IS3_Lm12ELm12EEEEENS1_IS3_Lm61ELm61EEEEENS1_IS3_Lm11ELm11EEEEENS1_IS3_Lm60ELm60EEEEENS1_IS3_Lm10ELm10EEEEENS1_IS3_Lm59ELm59EEEEENS1_IS3_Lm9ELm9EEEEENS1_IS3_Lm58ELm58EEEEENS1_IS3_Lm8ELm8EEEEENS1_IS3_Lm57ELm57EEEEENS1_IS3_Lm7ELm7EEEEENS1_IS3_Lm56ELm56EEEEENS1_IS3_Lm6ELm6EEEEENS1_IS3_Lm55ELm55EEEEENS1_IS3_Lm5ELm5EEEEENS1_IS3_Lm54ELm54EEEEENS1_IS3_Lm4ELm4EEEEENS1_IS3_Lm53ELm53EEEEENS1_IS3_Lm3ELm3EEEEENS1_IS3_Lm52ELm52EEEEENS1_IS3_Lm2ELm2EEEEENS1_IS3_Lm51ELm51EEEEENS1_IS3_Lm1ELm1EEEEENS1_IS3_Lm50ELm50EEEEENS1_IS3_Lm0ELm0EEEEENS1_IS3_Lm49ELm49EEEEaSERKNS2_ILm146EEE (rhs=..., this=0x7fffffffc4c0) at /path/to/cxxrtl.h:636
#4  _ZN6cxxrtl11concat_exprINS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS0_INS_10slice_exprINS_5valueILm273EEELm272ELm253EEENS1_INS2_ILm96EEELm63ELm37EEEEENS1_IS3_Lm98ELm98EEEEENS1_IS3_Lm48ELm48EEEEENS1_IS3_Lm97ELm97EEEEENS1_IS3_Lm47ELm47EEEEENS1_IS3_Lm96ELm96EEEEENS1_IS3_Lm46ELm46EEEEENS1_IS3_Lm95ELm95EEEEENS1_IS3_Lm45ELm45EEEEENS1_IS3_Lm94ELm94EEEEENS1_IS3_Lm44ELm44EEEEENS1_IS3_Lm93ELm93EEEEENS1_IS3_Lm43ELm43EEEEENS1_IS3_Lm92ELm92EEEEENS1_IS3_Lm42ELm42EEEEENS1_IS3_Lm91ELm91EEEEENS1_IS3_Lm41ELm41EEEEENS1_IS3_Lm90ELm90EEEEENS1_IS3_Lm40ELm40EEEEENS1_IS3_Lm89ELm89EEEEENS1_IS3_Lm39ELm39EEEEENS1_IS3_Lm88ELm88EEEEENS1_IS3_Lm38ELm38EEEEENS1_IS3_Lm87ELm87EEEEENS1_IS3_Lm37ELm37EEEEENS1_IS3_Lm86ELm86EEEEENS1_IS3_Lm36ELm36EEEEENS1_IS3_Lm85ELm85EEEEENS1_IS3_Lm35ELm35EEEEENS1_IS3_Lm84ELm84EEEEENS1_IS3_Lm34ELm34EEEEENS1_IS3_Lm83ELm83EEEEENS1_IS3_Lm33ELm33EEEEENS1_IS3_Lm82ELm82EEEEENS1_IS3_Lm32ELm32EEEEENS1_IS3_Lm81ELm81EEEEENS1_IS3_Lm31ELm31EEEEENS1_IS3_Lm80ELm80EEEEENS1_IS3_Lm30ELm30EEEEENS1_IS3_Lm79ELm79EEEEENS1_IS3_Lm29ELm29EEEEENS1_IS3_Lm78ELm78EEEEENS1_IS3_Lm28ELm28EEEEENS1_IS3_Lm77ELm77EEEEENS1_IS3_Lm27ELm27EEEEENS1_IS3_Lm76ELm76EEEEENS1_IS3_Lm26ELm26EEEEENS1_IS3_Lm75ELm75EEEEENS1_IS3_Lm25ELm25EEEEENS1_IS3_Lm74ELm74EEEEENS1_IS3_Lm24ELm24EEEEENS1_IS3_Lm73ELm73EEEEENS1_IS3_Lm23ELm23EEEEENS1_IS3_Lm72ELm72EEEEENS1_IS3_Lm22ELm22EEEEENS1_IS3_Lm71ELm71EEEEENS1_IS3_Lm21ELm21EEEEENS1_IS3_Lm70ELm70EEEEENS1_IS3_Lm20ELm20EEEEENS1_IS3_Lm69ELm69EEEEENS1_IS3_Lm19ELm19EEEEENS1_IS3_Lm68ELm68EEEEENS1_IS3_Lm18ELm18EEEEENS1_IS3_Lm67ELm67EEEEENS1_IS3_Lm17ELm17EEEEENS1_IS3_Lm66ELm66EEEEENS1_IS3_Lm16ELm16EEEEENS1_IS3_Lm65ELm65EEEEENS1_IS3_Lm15ELm15EEEEENS1_IS3_Lm64ELm64 [... cut ...]
  assign sig_10[252:175] = { out_data[63:36], 50'h3ffffffffffff };
  assign { sig_4[293], sig_4[175:174], sig_4[145:0] } = { sig_4[294], sig_4[296:295], sig_4[196:176], sig_4[296:295], sig_4[173:147], sig_4[292:197] };
  assign sig_5[226:174] = { out_data[36], 51'h7ffffffffffff, sig_10[174] };
  assign { out_data[35:32], out_data[0] } = { 4'hf, sig_11 };
endmodule
```

Thank you very much!
Flavien

### Expected Behavior

Successful completion of the simulation.

### Actual Behavior

Segmentation fault.

whitequark, 2024-01-07:
How'd you find all these? VLogHammer or a similar fuzzer?

flaviens, 2024-01-08:
Thank you for your response! Another fuzzing-based method.

whitequark, 2024-01-19:
Could you minimize the failure please? You can use the `bugpoint` pass to minimize segfaults.

whitequark, 2024-01-19:
I'm looking at this in the backtrace:

```
#0  0x000055555556d132 in cxxrtl::value<273ul>::blit<0ul, 0ul> (source=..., this=0x0) at /path/to/cxxrtl.h:254
```

This is suspicious to me. Why is `this=0x0`? This isn't ever OK to have in C++. But I also can't see any circumstance in which this would happen.

whitequark, 2024-01-19:
Cannot reproduce on commit fc5ff7a265293099e25812afbbbe8e687f0b62e8.

whitequark, 2024-01-20:
Unclear exactly what happened, but this was probably fixed sometime since Yosys 0.29. Please open a new issue if you can reproduce this on recent Yosys.

closed 2024-01-20 by whitequark